FreeCAD/FreeCAD #20185: PartDesign Multitransform Disabling Update View doesn't work.

leoheck, 2025-03-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have the `Update view` disabled but it is updating the view. Whats up with this feature?

Can we have this check box changed to a button to update the view?

[image]

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.21
  * DynamicData 2.73.0
  * Curves 0.6.59
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.5
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-03-13:
https://github.com/user-attachments/assets/90fc679c-716e-46cd-b4c4-41bf809a6a01

maxwxyz, 2025-03-19:
@kadet1090 maybe related to preview code?

leoheck, 2025-10-01:
Also, for all the transformations, if `Recompute on change` is disabled and a change happens, it should at least clear the old/outdated preview.