FreeCAD/FreeCAD #20187: [CAM] Integrate Better Tool Library workbench as default tool manager

phaseloop, 2025-03-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Let's face it - tool management in CAM workbench has the worst user experience of all features there.

We won't make anything even comparable to BTL in the coming years so we may as well integrate it and make it official.

https://github.com/knipknap/better-tool-library

What do you think?

@LarryWoestman @sliptonic 

### Full version info

```shell
1.0.0
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

LarryWoestman, 2025-03-14:
I don't know enough about this subject to have a valid opinion.

phaseloop, 2025-03-14:
I see that some of discussions were here:

https://github.com/FreeCAD/FreeCAD/issues/9466

It's really worth going back to this topic, AFAIK in 1.0.0 this is more instable and buggy than in previous versions.

Syres916, 2025-03-15:
I'm carrying out some testing on a variety of systems and will gather all my fixes in [this](https://github.com/knipknap/better-tool-library/compare/main...Syres916:better-tool-library:patch-1) branch, using non-default folders for macros/tool libraries is something that has to work across different platforms.

phaseloop, 2025-03-15:
Cool, thanks!

Thovthe, 2025-03-18:
Funny this was just posted this week; I'm sitting down to learn the CAM workbench with [this](https://blog.freecad.org/2023/11/24/tutorial-getting-started-with-the-path-workbench-part-one/) tutorial and the experience of trying to add tools to the library drove me to look for what I was doing wrong and this shows up.  Really almost anything would be better than what's there right now.  Also it may be broken in the flatpak because I can't seem to edit tools.

PR #21425 (2025-06-02): CAM: Integration of the tool and shape editors from BTL (Better Tool Library)
This very large PR essentially replaces CAM Tool Management completely, as part of the [Better Tool Library](https://github.com/knipknap/better-tool-library/) integration.

## Relevant Issues

Discussion: #9466

I also went through all open CAM issues to find any that may be resolved or otherwise affected:

- Resolves: #8201
- Resolves: #8526
- Resolves: #9625
- Resolves: #11522
- Resolves: #12712
- Likely resolves: #12823, but should be tested to be sure
- Possibly resolves #13603, to be tested
- Resolves: #17083
-  [... cut]