FreeCAD/FreeCAD #20189: nativeIFC  utf-8 properties  will not be showed and edited

wenwan1, 2025-03-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

property Group name  messed up  and other utf-8 character propertyName is hidden and not support for edit
[image]

### Full version info

```shell
Freecad 1.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wenwan1, 2025-03-14:
when show ifc entity properties will preprocess to ascii  and full utf-8 property name was refused 
[image]

will throw error "invalid property name ..."  can any ways to edit utf-8 property Name ?
[image]

yorikvanhavre, 2025-03-15:
Indeed up to now property names in FreeCAD must be ASCII. This is not BIM-specific.

This is indeed something that should be implemented. Or directly, or with the same system as other structures in FreeCAD (a name (ASCII) and a label (UTF-8).

yorikvanhavre, 2025-03-20:
The file manager is not really part of FreeCAD, it's a Qt functionality. What's the problem there?

kaiwas, 2025-03-20:
> What's the problem there?

When I open the file I see this.
Moreover, all paths are without Cyrillic.

[image]

yorikvanhavre, 2025-03-20:
Can't reproduce... Maybe something not configured correctly on your system? Or maybe an appimage problem?

[image]

```
OS: Debian GNU/Linux trixie/sid (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.40675 (Git)
Build date: 2025/03/11 12:58:03
Build type: Unknown
Branch: main
Hash: 61eedbddc4dfc9416995262e4ad182078bfbab02
Python 3.13.2, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1
Installed mods: 
  * dxf-library
  * CfdOF 1.29.5
  * OpenTheme 2024.9.1
  * parts_library
  * freecad.gears 1.3.0
  * OpenDark 2024.3.13
  * CADExchanger
  * ExtremeProDark 2.9.0
```

kaiwas, 2025-03-20:
> Can't reproduce... Maybe something not configured correctly on your system? Or maybe an appimage problem?

I don't know. Maybe.
I'll probably have to file an issue on this topic. Since it's not related to the issue discussed here. Thanks!