FreeCAD/FreeCAD #20190: DLL load failed while importing _coin

yan695432, 2025-03-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When running the released exe using the source code of version 1.0.0, it prompts "DLL load failed while importing _coin"

[image]

### Full version info

```shell
version:1.0.0
libPack:LibPack-1.0.0-v3.0.0
Visual Studio Code:1.98.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-21:
does it also happen with 1.1?

FEA-eng, 2025-10-01:
@yan695432 Is it still an issue with your self-compiled Windows builds?

luzpaz, 2025-11-23:
Closing as no response from OP