FreeCAD/FreeCAD #20192: OpenSCAD import file crash

prime6809, 2025-03-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Open a new document
Select OpenSCAD workbench 
Select File->import
Import the scad file : https://github.com/skiselev/micro_8088_case/blob/master/printed/Micro_8088_Case_Modular.scad
Nothing is drawn and crash reported in report view : 

`09:39:18  [<Part::PartFeature>]
09:39:18  MultMatrix
09:39:18  Multmatrix
09:39:18  [['1', '0', '0', '3'], ['0', '1', '0', '3'], ['0', '0', '1', '3'], ['0', '0', '0', '1']]
09:39:18  Matrix ((1,0,0,3),(0,1,0,3),(0,0,1,3),(0,0,0,1))
09:39:18  Apply Multmatrix
09:39:18  Fuse
09:39:18  [<Part::PartFeature>, <Part::PartFeature>]
09:39:18  Single Fuse
09:39:18  Check Object Shape
09:39:18  Shape is Null - recompute
09:39:18  Check Object Shape
09:39:18  Shape is Null - recompute
09:39:18  Recompute failed : difference010
09:39:18  pyException: Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "D:\apps\FreeCAD-weekly\bin\Lib\site-packages\freecad\module_io.py", line 16, in OpenInsertObject
    getattr(importerModule, importMethod)(objectPath, *importArgs, **importKwargs)
  File "D:\apps\FreeCAD-weekly\Mod\OpenSCAD\importCSG.py", line 145, in insert
    processcsg(tmpfile)
  File "D:\apps\FreeCAD-weekly\Mod\OpenSCAD\importCSG.py", line 174, in processcsg
    result = parser.parse(f.read())
             ^^^^^^^^^^^^^^^^^^^^^^
  File "D:\apps\FreeCAD-weekly\bin\Lib\site-packages\ply\yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\apps\FreeCAD-weekly\bin\Lib\site-packages\ply\yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "D:\apps\FreeCAD-weekly\Mod\OpenSCAD\importCSG.py", line 979, in p_multmatrix_action
    part = fuse(p[6],"Matrix Union")
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\apps\FreeCAD-weekly\Mod\OpenSCAD\importCSG.py", line 593, in fuse
    myfuse.Shape = myfuse.Base.Shape.fuse(myfuse.Tool.Shape)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: {'sclassname': 'class Part::NullShapeException', 'sErrMsg': 'Null input shape', 'sfile': 'C:\\bld\\freecad_1741027165263\\work\\src\\Mod\\Part\\App\\TopoShapeExpansion.cpp', 'iline': 5663, 'sfunction': 'class Part::TopoShape &__cdecl Part::TopoShape::makeElementBoolean(const char *,const class std::vector<class Part::TopoShape,class std::allocator<class Part::TopoShape> > &,const char *,double)', 'swhat': 'Null input shape', 'btranslatable': False, 'breported': False}
`

Cheers.

Phill.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Logical/physical DPI: 96/105.985
Installed mods: 
  * 3D_Printing_Tools
  * Alternate_OpenSCAD 1.0.0
  * Curves 0.6.58
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * MeshRemodel 1.10.36
  * pcb 6.2023.1
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * ThreadProfile 1.92.0
  * toSketch 1.0.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

prime6809, 2025-03-16:
Further datapoint, above linked scad file contains several model, I have exported each one to a .csg, and only one of them causes a crash when importing, the front of the case, which I will attach here.

[Front.zip](https://github.com/user-attachments/files/19270576/Front.zip)

With the current developer release : importing using the New importer causes freecad to crash to the desktop, using the old importer causes the importer to crash with the following report in the Report view : 

`10:01:33  Vector
10:01:33  Vector
10:01:33  Matrix
10:01:33  ('height', '0.3')
10:01:33  ('$fn', '90')
10:01:33  ('$fa', '12')
10:01:33  ('$fs', '2')
10:01:33  ('text', 'Micro 8088')
10:01:33  ('size', '8')
10:01:33  ('spacing', '1')
10:01:33  ('font', 'Bahnschrift')
10:01:33  ('direction', 'ltr')
10:01:33  ('language', 'en')
10:01:33  ('script', 'Latn')
10:01:33  ('halign', 'center')
10:01:33  ('valign', 'default')
10:01:33  ('$fn', '90')
10:01:33  ('$fa', '12')
10:01:33  ('$fs', '2')
10:01:33  textmsg : text ( text="Micro 8088", size = 8, spacing = "1", font = "Bahnschrift", direction = "ltr", language = "en", script = "Latn", halign = "center", valign = "default", $fn = 90, $fa = 12, $fs = 2);
10:01:33  Unknown value for the valign parameter (use "baseline", "bottom", "top" or "center"): 'default'
Geometries in cache: 1
Geometry cache size in bytes: 39232
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.009
   Top level object is a 2D object:
   Contours:       18

10:01:33  pyException: Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "D:\apps\FreeCAD-weekly\bin\Lib\site-packages\freecad\module_io.py", line 16, in OpenInsertObject
    getattr(importerModule, importMethod)(objectPath, *importArgs, **importKwargs)
  File "D:\apps\FreeCAD-weekly\Mod\OpenSCAD\importCSG.py", line 127, in open
    processcsg(filename)
  File "D:\apps\FreeCAD-weekly\Mod\OpenSCAD\importCSG.py", line 174, in processcsg
    result = parser.parse(f.read())
             ^^^^^^^^^^^^^^^^^^^^^^
  File "D:\apps\FreeCAD-weekly\bin\Lib\site-packages\ply\yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\apps\FreeCAD-weekly\bin\Lib\site-packages\ply\yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "D:\apps\FreeCAD-weekly\Mod\OpenSCAD\importCSG.py", line 1271, in p_text_action
    p[0] = [processTextCmd(t)]
            ^^^^^^^^^^^^^^^^^
  File "D:\apps\FreeCAD-weekly\Mod\OpenSCAD\importCSG.py", line 916, in processTextCmd
    from OpenSCAD2Dgeom import importDXFface
<class 'ImportError'>: cannot import name 'importDXFface' from 'OpenSCAD2Dgeom' (C:\Users\afra\AppData\Roaming\FreeCAD\Mod\pcb\.\OpenSCAD2Dgeom.py)
`

Seems to be related to the text on the front (bottom) of the model

Cheers.

Phill.

FEA-eng, 2025-12-14:
See [this](https://github.com/FreeCAD/FreeCAD/issues/20192#issuecomment-3355484219)

It's not even in the supported formats.

KeithSloan, 2026-04-18:
My understanding is the internal FreeCAD - OpenSCAD workbench will be dropped in FreeCAD 1.2

I plan to add the new External Workbench to Addon Manager. as soon as I am more confident that it works better than
the Alternate Importer. It still lacks certain functionality.

As the the new importer is AST based then it should be able to address issues with booleans that will probably never be fixed in the Alternate Importer.

FEA-eng, 2026-10-19:
No crash, but still cannot import.

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 617.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```