FreeCAD/FreeCAD #20193: Draft_Edit does not show popup menu when clicking while holding 'E' key

semhustej, 2025-03-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open new file
- create a Draft_Wire with several segments
- select the Draft_Wire and enter `Draft_Edit` mode
- hold the `E` key and click on an edge or node
- a popup menu should be shown, but this does not happen

I could reproduce in FreeCAD 1.0. Apparently it works in FreeCAD 0.21.2 (as per this forum thread https://forum.freecad.org/viewtopic.php?t=95581), so it seems like a regression.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

Draft

### Anything else?

As a workaround, user can use the `Alt` key instead of `E`.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-03-14:
This is due to the `E` being used as a shortcut in other workbenches (equal constraint in Sketcher, explode in Assembly). If your start up workbench is Draft, and you have not auto-loaded another workbench, the problem does not occur. It may be possible to avoid the problem by switching from `coin.SoKeyboardEvent.DOWN` to `coin.SoKeyboardEvent.UP` however.

semhustej, 2025-03-15:
If it is problematic to solve, I think it would be better to just remove the `E` key shortcut, as incosistencies are very confusing for users. I think having one key (Alt key) should be enough.

adrianinsaval, 2025-03-17:
> I think having one key (Alt key) should be enough.

I think alt is more likely to have conflicts

closed 2025-03-17 by chennes

semhustej, 2025-03-27:
> > I think having one key (Alt key) should be enough.
> 
> I think alt is more likely to have conflicts

Alt key currently works the same way as the E key. User has option to use both.

PR #20201 (2025-03-17): Draft: make Draft_Edit also detect key up events
Fixes #20193.

The key down events can be claimed by shortcuts from other workbenches. Detecting both the down and up events does not seem to cause problems.