FreeCAD/FreeCAD #20195: ImportDXF in CLI RuntimeError

vl-201, 2025-03-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am trying to import a DXF in the CLI, however I am getting the following error:
```
Traceback (most recent call last):
  File "/Users/vl/projects/process_dxf_file.py", line 91, in <module>
    importDXF.insert(input_file, doc_name)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/importDXF.py", line 2880, in insert
    Import.readDXF(filename)
RuntimeError: <built-in function _getframe> returned a result with an exception set
```

### Full version info

```shell
OS: macOS 15.3.2
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Anything else?

I run FreeCAD in terminal with my script: `python process_dxf_file.py input_file.dxf`

This is how my script looks like:
```
import sys
import json

sys.path.append("/Applications/FreeCAD.app/Contents/Resources/lib/")

import FreeCAD
import importDXF

def process_obj(doc):
    """
    some object manipulation
    """

if __name__ == "__main__":
    input_file = sys.argv[1]
    doc_name = "Unnamed"

    importDXF.insert(input_file, doc_name)
    FreeCAD.setActiveDocument(doc_name)
    doc = FreeCAD.getDocument(doc_name)

    processed_obj = process_obj(doc)
    print("result:", json.dumps(processed_obj))
```

This script has not been changed in any way and worked fine in version 0.21.2, however, after the update, DXF files are no longer imported

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-16:
Could you check if the issue is present in the 1.1 dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Does it happen with every DXF and could you share one?

vl-201, 2025-03-17:
> Could you check if the issue is present in the 1.1 dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds Does it happen with every DXF and could you share one?

I had no luck installing the dev version on my mac because the package is damaged (I tried downloading it several times, but the result is the same)
[image]

However, I was able to try the dev version on a PC with ubuntu 22.04
There the error is still the same:
```
No INSUNITS or MEASUREMENT; setting scaling to 1 DXF unit = 1mm based on DXF scaling option
First reference to missing Layer 'Layer 1'Traceback (most recent call last):
  File “<string>”, line 1, in <module>
  File “/tmp/.mount_FreeCACDBNiA/usr/Mod/Draft/importDXF.py”, line 2835, in open
    Import.readDXF(filename)
<class 'RuntimeError'>: <function _import at 0x7fd13b5bb060> returned a result with an exception set
Exception while processing file: ../dxf_input.dxf [<function _import at 0x7fd13b5bb060> returned a result with an exception set].
```

I've tried several different DXF files
I've attached them below

[test_dxf_files.zip](https://github.com/user-attachments/files/19291295/test_dxf_files.zip)

yorikvanhavre, 2025-03-18:
@kpemartin any idea of what's happening here?

kpemartin, 2025-03-18:
@vl-201 Since I can't seem to reproduce the additional error messages, here's an experiment to try, to see whether or not the missing layer declaration is the source of the problem:

Use a text editor to correct the DXF file (or, if you have access to whatever generates them, change the generating code).
In the LAYER table of the TABLES section, for instance at line 69 in dxf_input.dxf, the file reads:
```
  0
TABLE
  2
LAYER
 70
1
  0
LAYER
  2
0
```
The last (type 2) record (pair of lines) is the name `0` of the declared layer; this layer is not used by the document. Change this line from `0` to `Layer 1`, save the file and try importing it to see if you still get the traceback and exception messages.

vl-201, 2025-03-18:
@kpemartin 
Thank you.
I tried changing 0 to Layer 1, and running the script on both my OS, but the result is the same.
Here is the traceback from the macOS:
```
Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/importDXF.py", line 2877, in insert
    import ImportGui
  File "/Users/vl/projects/anaconda3/envs/env/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ImportError: Cannot load Gui module in console application.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/vl/projects/process_dxf_file.py", line 20, in <module>
    importDXF.insert(input_file, doc_name)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/importDXF.py", line 2880, in insert
    Import.readDXF(filename)
RuntimeError: <built-in function _getframe> returned a result with an exception set
```

Here is the traceback from the ubuntu 22.04:
```
No INSUNITS or MEASUREMENT; setting scaling to 1 DXF unit = 1mm based on DXF scaling option
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/tmp/.mount_FreeCAOncNbn/usr/Mod/Draft/importDXF.py", line 2835, in open
    Import.readDXF(filename)
<class 'RuntimeError'>: <function _import at 0x7f77b7e9f060> returned a result with an exception set
Exception while processing file: dxf_input.dxf [<function _import at 0x7f77b7e9f060> returned a result with an exception set]
```

[dxf_input.zip](https://github.com/user-attachments/files/19321827/dxf_input.zip)

kpemartin, 2025-03-19:
Silly me, I was running your macro code from within the Gui. Perhaps the importer has some confusion as to whether the Gui is available or not. I'll look further tomorrow.

kpemartin, 2025-03-19:
After fighting with the debugger to see the original exception, I found the problem.
I had code checking if the layer's View object existed by looking for a null pointer rather than `Py_None`, so it tried setting properties like `OverrideLineColorChildren` on the `None` object. This would leave an exception on the current state of the python engine. This acts as a booby trap that much further down the line will cause some unrelated call to fail with a bogus `(something) returned a result with an exception set` SystemException. The original exception (an AttributeException that specifically names the attribute) is in the `cause` of this exception, but nothing seems to show that, so I have to spend ages in the debugger digging around (including remembering how the shovel works) to get the information. VS is supposed to have a dual python/native debugging mode that might have helped me but I can't seem to get it to work.

Because of the try/except block that was used to determine if the Gui or non-gui importer should be used, this added another confusing layer to the stack traceback. This particular bit of code now just checks the `gui` flag (duh!) to decide which importer to use.

PR on its way after I run the unit tests.

It has nothing to do with the undeclared layer in the DXF file.

kpemartin, 2025-03-20:
PR #20328 submitted including a unit test to detect the condition and a fix.

vl-201, 2025-03-21:
@kpemartin Wow, thank you so much for the very quick fix for this problem!

Do you happen to have any idea when this PR will be moderated?

kpemartin, 2025-03-23:
@vl-201 I'm afraid that I haven't been here long enough to have any idea of how the internal machinery works. So far it seems to me that one of maybe 3 or 4 people whose handles I see all the time assigns it to someone I don't know for review. Once it passes review it gets put in a queue for inclusion. But I really don't know how many people are actually involved in FreeCAD development nor how they're organized in terms of responsibility and (for lack of a better term) power. So far I think I've had a different reviewer every time.

I expect how long it takes depends on workload and perceived importance.

I believe that if you can build it yourself you can fetch my version to try it out. I  don't recall the exact incantation but the URL `https://github.com/kpemartin/FreeCAD/tree/Issue20195` might work.

closed 2025-03-31 by chennes