FreeCAD/FreeCAD #20203: "Create projection on surface" appears to be projecting incorrectly

Cinderella-Man, 2025-03-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello,

I'm trying to project face on curved object and the resulting projection appears to be "weird". I believe this is a regression.

I created a project that would showcase the behaviour. The attached zip file contains two files.

First one is before applying the projection. It contains curved surface and sketch.

Here's the object and sketch next to each other:
[image]

The view which way projection should happen:
[image]

Now, I'm converting the sketch to face using Part -> "Make face from wires":

[image]

I'm clicking on "Create projection on surface" and "select projection surface":

[image]

clicking on "add face" and selecting the face created from the sketch:

[image]

Now I move camera to "Right":

[image]

and clicking on the "Get current camera direction" button. "OK" button to create the projection.

The result is as follows (after hiding the object and face):

[image]

I expected a resulting projection to have very similar shape to the original object sharp edges on the bottom etc.

Second file contains the same project as the first one but with resulting projection

[00_sample.zip](https://github.com/user-attachments/files/19254488/00_sample.zip)

PS. I have another project which is very similar to the attached one which is breaking in a different way but I prefer not to share the project publicly.

Please let me know is this me doing something wrong or it's an regression (as I believe that I was doing this before and it was working)

Thanks in advance,

Kamil

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FEA-eng, 2025-10-01:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2025-12-14:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251209 (Git shallow)
Build date: 2025/12/09 13:53:17
Build type: Release
Branch: grafted,grafted
Hash: 5ad6a1ba58c665c23f1d217c23e74379d8c1a5a4
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```