FreeCAD/FreeCAD #20205: PartDesign: Feature suppression results in inappropriate changes to Placement for some Features

jspiros, 2025-03-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

_I originally wrote this as a comment on #6164, but upon spending the day familiarizing myself with the relevant FreeCAD code, I am reasonably confident that this is a unique bug specifically brought on by the Suppression feature._

### What's happening

I'm trying to use a Fasteners workbench ScrewTap as the tool body to cut a thread into a PartDesign Body with PartDesign Boolean. The PartDesign Body in question is built, in part, using an AdditivePipe.

When I create the Boolean, it works as expected, it cuts and everything is oriented correctly. As I want this threaded hole to be optional per parameters, I tested setting the Suppressed property of that Boolean feature to true, and then back to false. Upon recompute after un-suppressing, the Boolean appears with the entire PartDesign Body's solid rotated 90 degrees about the X axis (such that what was the "Top" face in the previous feature is now the "Front" face). It still performs the cut, and the cut at least looks correct, but this breaks every feature after the Boolean (because nothing is where it's supposed to be anymore).

If I suppress again, it does restore the correct rotation, but any un-suppression of that Boolean feature going forward rotates as before. I can only restore the initial correct cut by deleting and recreating the Boolean feature, and never suppressing it.

### Reproducing and testing the bug

Trying to reduce this to the simplest case, I created a test PartDesign Body made from a single AdditivePipe, where the path sketch is on the XY plane and the profile sketch is on the XZ plane, and I used a Part primitive as the Boolean tool body. This is reproducible in that case as well.

It is also reproducible if the PartDesign Body is built using an AdditiveLoft, with the same sketch on the XZ plane as the previous test but a different sketch on the XY plane to act as the other profile.

It is also reproducible if the PartDesign Body is built, using the same XZ plane sketch as the previous two tests, with a Revolution that revolves around the Z axis. **HOWEVER**, it is **not** reproducible if the PartDesign Body is built with a Revolution that revolves around the X or Y axes, using the same XY plane sketch from the AdditiveLoft test.

I also tested with Fuse instead of Cut, and that behaves identically: the same thing where it works fine until it has gone through a suppression cycle. Contrary to bug #6164, **this happens regardless of whether the tool body or the PartDesign Body have adjusted placements beforehand**. Even if the tool body is non-overlapping and I set it to cut, such that the operation does effectively nothing in terms of final result, it still rotates the body after a suppression cycle, in all of the cases where it's reproducible.

### Investigation into the cause

After briefly skimming the code, and testing in the Python console, I was able to determine that this only happens when a PartDesign Body already a feature that, as part of its normal functioning, has a different placement than the default/identity placement. Features like Revolution, AdditivePipe, and AdditiveLoft, can end up with a placement with a 90 degree rotation, despite looking normal and correct in the end.

Some later feature types, such as Pads, will then match that placement change and carry it forward, adopting it for themselves. But, other features, such as Booleans—and I later confirmed Additive/SubtractivePrimitives too, at least, and perhaps others—do not carry that placement forward, and instead change it back to the default/identity placement, when functioning properly.

So, for a Body made up of a Revolution, Pad, and Boolean, this is what you get when it's rendering properly:

```
>>> App.getDocument("Unnamed").getObject("Body").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument("Unnamed").getObject("Revolution").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument("Unnamed").getObject("Pad").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument("Unnamed").getObject("Boolean").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
```

However, after a suppression cycle, this changes to:

```
>>> App.getDocument('Unnamed').getObject('Body').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Revolution').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument('Unnamed').getObject('Pad').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument('Unnamed').getObject('Boolean').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
```

After spending the last day and a half looking into the code, I've determined that this is almost certainly caused by the suppression-related function below, especially line 142 when it sets the shape for the feature using the shape of the previous feature:

https://github.com/FreeCAD/FreeCAD/blob/9a6a98f824436656cae242ae26b043f78411ebba/src/Mod/PartDesign/App/Feature.cpp#L122-L144

I am able to reproduce the rotation behavior without touching the suppression code at all, by using the Python console to set the shape of the Boolean feature to the shape of its predecessor feature, and then recomputing (which should recreate the intended shape, and does, but rotated as the placement of the previous shape is retained):

```
App.getDocument('Unnamed').getObject('Boolean').Shape = App.getDocument('Unnamed').getObject('Pad').Shape
App.getDocument('Unnamed').getObject('Boolean').recompute()
```

And I can restore the correct rotation by clearing the Placement and recomputing:

```
App.getDocument('Unnamed').getObject('Boolean').Placement = App.Placement()
App.getDocument('Unnamed').getObject('Boolean').recompute()
```

I'm not sure why it only works in one way, remembering the placement from the previous feature's shape and imposing it on the recomputed shape when recomputing assigns the correct shape, but the original computed shape's placement isn't imposed on the previous feature's shape when that one's assigned.

As for how it works in the former case at least, my best guess is that this onChanged code might be what's doing it, though I don't see why it would only run/detect a change in one direction:

https://github.com/FreeCAD/FreeCAD/blob/9a6a98f824436656cae242ae26b043f78411ebba/src/Mod/Part/App/PartFeature.cpp#L1474-L1509

### Workaround

For now, I'm working around this bug by showing hidden properties, and setting an expression of `.SuppressedShape.Placement` for the Placement property of the Boolean, or any other similarly affected feature. I do notice that despite that expression, while suppressed, it still obtains the updated Placement from the new shape, and doesn't evaluate the expression. But, upon un-suppression it recomputes Placement using the expression and everything works properly, and reliably, as I hoped.

This workaround takes advantage of the fact that the SuppressedShape never adopts the updated Placement (as the code that generates the SuppressedShape starts with the identity Placement explicitly, perhaps), which is maybe also why no problems occur with the references that it apparently exists to preserve (per https://github.com/FreeCAD/FreeCAD/issues/19308#issuecomment-2627055865)?

However, I'm not sure if the SuppressedShape should be part of the eventual solution to this problem in the code itself. In the same Revolution -> Pad -> Boolean example, if the Pad is suppressed, the Pad's SuppressedShape also has the default/identity Placement, not the correct-for-that-feature Placement that's rotated, so always restoring that upon un-suppression wouldn't be appropriate. Though, placement is ReadOnly and automatically calculated for the Pad (and many other features) when there's a profile assigned thanks to use of `ProfileBased::positionByPrevious`. So perhaps that's the required differentiator?

If no one else has time, I might be willing to try to figure out how to fix this and submit a pull request, but, I'm a bit insecure as I don't feel confident that I know why many of these things are implemented this way in the first place, at least not enough to know the best way or place to address it.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.0.0.
Build type: Release
Python 3.13.1, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * Curves 0.6.58
  * dxf-library
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-21:
@FlachyJoe FYI

FlachyJoe, 2025-04-11:
@jspiros I can not reproduce. Can you please share a failing file?

jspiros, 2025-04-13:
@FlachyJoe Wonderful, thank you very much for finding the time to look into this and fix it!

> FYI I follow the problem to these lines

> Setting the `Shape` in `Feature::updateSuppressedShape` trigger a `Placement` update from the `Shape` transform.
> I'm still working on the fix.

And thank you for sharing your analysis here. I appreciate the confirmation that I was on the right track in identifying what was going on and in coming to the same line in `Feature::updateSuppressedShape` triggering that `Feature::onChanged` code. I had also thought that keeping and restoring the original Placement might be the simplest and safest solution, so it's nice to also have that intuition confirmed, and to be able to see and learn from where and how exactly you implemented it. 😄

PR #20770 (2025-04-16): PartDesign: Fix #20205 where Placement changes at Suppression
Fix #20205

Keep trace of the original placement and restore it when the feature is unsuppressed.
commit 57b9a413 (2025-04-16): PartDesign: Fix #20205 where Placement changes at Suppression
closed 2025-04-16 by kadet1090