FreeCAD/FreeCAD #20208: Sketcher: Creating Ellips by centre reults in crash

brdi, 2025-03-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Reproducable  error:

New file; Sketcher; New Sketch Create Ellips by Centre; Centrepoint on 0,0 (coincident), Top on Z-Axe -(constraint point on object) -------> Crash.
See attached problem report from (MacBook Apple M2 Max on Sequoia 15.4 and 64 GB internal mem)

[foutfreecad.txt](https://github.com/user-attachments/files/19262893/foutfreecad.txt)

### Full version info

```shell
OS: macOS 15.4
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Freecad-Built-in-themes-beta 1.2.2
  * FreeCAD-themes 2025.1.7
  * OpenTheme 2024.9.1
  * cadquery-freecad-workbench-2.0.0 2.0.0
  * fasteners 0.5.37
```

### Subproject(s) affected?

None

### Anything else?

This error appeared when I followed Mango Jellys  instruction on
https://youtu.be/IALkQrtdJcg?t=519
 I then tried to reproduce it and I found out that creating an ellips is not the problem
The crash seems to happen when FreeCAD tries to obtain information about a specific point on the ellipse during constraint placement. This problem is probably caused by a bug in the Sketcher module in combination with the way the ellipse is created and edited.

Related messages:
SketcherGui::DrawSketchDefaultHandler<SketcherGui::DrawSketchHandlerEllipse, SketcherGui::StateMachines::ThreeSeekEnd, 3, SketcherGui::ConstructionMethods::CircleEllipseConstructionMethod>::getPointInfo

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

brdi, 2025-03-15:
In Safemode the error appeared as well
Find attached the apple problem report

[foutFreeCadSafeMode.txt](https://github.com/user-attachments/files/19263007/foutFreeCadSafeMode.txt)

fmckeogh, 2025-04-24:
Also experiencing this bug

luzpaz, 2025-05-07:
@johnocfii do you mind reproducing on macOS ?

luzpaz, 2025-05-07:
@brdi is there a way you can also upload the model (by zipping it up) and attaching to this ticket ?

JohnOCFII, 2025-05-07:
> [@JohnOCFII](https://github.com/JohnOCFII) do you mind reproducing on macOS ?

I'll try to get to it.  But both of the reports were with the released version of 1.0.  I seem to recall some work in that area.  If would be good if they could test with a weekly development build. https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.05.06

brdi, 2025-05-07:
I’ll look into it later this evening or tomorrowbrdi!- mobiel verzonden, typos may occurr -!

brdi, 2025-05-10:
@luzpaz 

No use to ipload a model. There is/was no model. I just start with a new model. The first thing I do is create a sketch in the XZ-plane. In the sketch I create an ellips by centre, start the ellips with the centre on 0,0 (my autocostraints are so that the centre has a point-to-point relation with the origin (0,0) of the sketch.  When I try to snap the top of the ellips to the Z-axis, the error occurs. So I don't know what to upload. Before I create the ellips, the model only contains an empty sketch (XZ-plane)

However, this morning I tried to reproduce the error on my iMAC (Intel based) and there it did not happen (using the 1.0 official release Revnr. 39109 dated 2024/11/18 16:48:00 on MacOS 15.4.1 z86_64 Hash 2fcc531).
I'll give it a shot on my MacBook that has an ARM architecture and that is where the original error occurred.

luzpaz, 2025-05-10:
@brdi upload the screencast somewhere else and link to here ?

brdi, 2025-05-10:
In the weekly build (1.1.0dev / 41628   2025/05/09 10:50:32 Hash 9577d10) I noticed the following:

1.  "About Freecad" gives as Architecture  x86_64. Weird (?). I am running it on  a arm64 based Apple M2 Max and not on an Intel based Mac or Windows machine.
2.  When trying to recreate the error: The ellips will not snap automatically to the Z-axe when I create it, how hard I try. Hence the error can not be reproduced. Not since it is solved but since some functionallity is lacking. I have to ad the constraint manually by selecting the ellips top and the Z-axe and then add the constraint (linme2point). No error is fine by me but I think that the root cause is not solved. FreeCad does not attempt (appertly) to dd the autoconstraint and therefore the error is prevented.
3.  In Freecad 1.0  it is perfectly possible to create an ellipse by centre al;s long as you do not use the autoconstraintfunction when creating the ellips. The addition of the contraint as described above is the work around in the 1.0 version to avoid the error.

Hope this more or less helps to find the root cause.

brdi, 2025-05-10:
> [@brdi](https://github.com/brdi) upload the screencast somewhere else and link to here ?
@luzpaz 
Screencast can be found here:
https://1drv.ms/u/c/2cbeb6bd236d1312/EUYTEQOQoCVLv73qtr1USWEB_7qAbr5fkUcHq-U0H_Hr0A?e=H7r9vB

brdi, 2025-05-10:
https://1drv.ms/v/c/2cbeb6bd236d1312/EdVycvSUEBhNuuwOPsGxpEMBscKwDi_cXqDo0I7DAJYTSQ?e=zWFSK5

luzpaz, 2025-05-11:
> 1.  "About Freecad" gives as Architecture  x86_64. Weird (?). I am running it on  a arm64 based Apple M2 Max and not on an Intel based Mac or Windows machine.

@brdi did you install the arm64 or x86 build ? (there are 2 builds, can you look at the filename? )

Note: I added your screencast (downsampled) to the OP, thanks for that.

FEA-eng, 2025-10-02:
@maxwxyz Can you check if it still happens on Mac?

brdi, 2025-10-02:
> FEA-eng
>  left a comment
> (FreeCAD/FreeCAD#20208)
> @maxwxyz <https://github.com/maxwxyz> Can you check if it still happens on Mac?

Issue can be closed.
I thought it was closed allready.

I Could not recreate error today.

brdi