FreeCAD/FreeCAD #20212: SIGSEGV crash on save

jonhnet, 2025-03-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Open a document, then click File->Save, and FreeCAD crashes with a SIGSEGV. No crash in 1.0.0, new in weekly-builds-40504.

Repro steps:

Unpack the attached .tgz
Open in FreeCAD the file fom2-frame-beams.FCStd. (It will open dependent files.)
Click File->Save

Observed: crash with SIGSEGV
Expected: save the file, no crash

Stack trace:

[stack_trace.txt](https://github.com/user-attachments/files/19265018/stack_trace.txt)

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/109.201
Installed mods: 
  * A2plus 0.4.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-03-16:
I cannot reproduce the crash with `FreeCAD_weekly-builds-40317-conda-Linux-x86_64-py311.AppImage`, while I can with the newer version. This is caused by a very recent change. 

It feels like the bug has been here for a long time, a recent commit simply uncovered it.

xtemp09, 2025-03-16:
@tritao, this commit 006feaf276db119ec01071aa72c5c2300d84f0f6 (PR #19900) introduces the crash. Possibly, it just uncovers existing bug.

PR #20223 (2025-03-19): [App] Fix the crash reported in #20212
#19900 has a typo, this commit fixes it. Closes #20212.

CC @tritao
closed 2025-03-19 by adrianinsaval