FreeCAD/FreeCAD #20213: Core: Line and Point color still use RGBT instead of RGBA

Roy-043, 2025-03-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The Diffuse color used for faces has been changed from RGBT to RGBA (transparency to alpha). But the color used for lines and points still seems to use RGBT.

Test:
Paste the code below in the Python console.
```python
doc = App.ActiveDocument
obj = doc.addObject("Part::Box", "Box")
print(obj.ViewObject.LineColor)
print(obj.ViewObject.PointColor)
print(obj.ViewObject.ShapeAppearance[0].DiffuseColor)
```

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

benj5378, 2025-03-17:
@bofdahof could you be interested in helping us with this issue?

benj5378, 2025-03-29:
```py
>>> doc = App.ActiveDocument
>>> obj = doc.addObject("Part::Box", "Box")
>>> print(obj.ViewObject.LineColor)
(0.09803921729326248, 0.09803921729326248, 0.09803921729326248, 1.0)
>>> print(obj.ViewObject.PointColor)
(0.09803921729326248, 0.09803921729326248, 0.09803921729326248, 1.0)
>>> print(obj.ViewObject.ShapeAppearance[0].DiffuseColor)
(0.800000011920929, 0.800000011920929, 0.800000011920929, 1.0)
```

I cannot reproduce; has the issue already been fixed?

Roy-043, 2025-03-30:
Oops, you are completely right. I see that I tested with V1.0 instead of with a recent dev version. Sorry for the confusion. 😓

closed 2025-03-30 by Roy-043