FreeCAD/FreeCAD #20221: CAM Custom Post-Processing Script is incompatible with conda packaging

jalapenopuzzle, 2025-03-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am using FreeCAD 1.0.0 on Linux using the conda package.
I am attempting to add a custom CAM post-processing script.

The path to add the script has changed from earlier versions due to renaming the Path workbench to CAM. This has also caused the ~/.local/share/FreeCAD/Mod/Path directory to be renamed to ~/.local/share/FreeCAD/Mod/CAM, and custom post-processing scripts appear to belong at ~/.local/share/FreeCAD/Mod/CAM/Path/Post/scripts .
Checking FreeCAD's internal search path in the python console:

```
>>> import Path.Post
>>> print(Path.Post.__path__)  # Should show all directories being searched
['/tmp/.mount_FreeCACGdopf/usr/Mod/CAM/Path/Post']
>>> 
```

This shows that FreeCAD is ONLY searching the internal conda package path, and is not searching the user's config for custom post-processing scripts.
Furthermore, creating the directory ~/.local/share/FreeCAD/Mod/CAM prevents the CAM workbench from loading at all when FreeCAD is started.

I will attempt to work around this issue and continue my development by building FreeCAD from source.
However, this issue makes it impossible to distribute the completed post-processing plugin except by merging it into FreeCAD (which I intend to do when ready).

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * freecad.gears 1.3.0
  * ThreadProfile 1.93.0
  * fasteners 0.5.37
  * btl 0.9.9
  * 3D_Printing_Tools
  * sheetmetal 0.7.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

phaseloop, 2025-03-16:
I'll be able to check it later during week, but in the meantime:

- do you see any error when the workbench fails to load?
-  this theoretically makes sense so if the directory does not exists it is not being added to the search paths
- did you only add Mod/CAM directory? What happens if you would add full Mod/Cam/Path/Post/scripts?

jalapenopuzzle, 2025-03-16:
Regarding your questions:
I created the following path:

- ~/.local/share/FreeCAD/Mod/CAM/CAM/Post/scripts -> CAM workbench missing
- ~/.local/share/FreeCAD/Mod/CAM/Path/Post/scripts -> CAM workbench missing
- ~/.local/share/FreeCAD/Mod/CAM  -> CAM workbench missing

In all cases above, no error messages were displayed:

```
user@host:/path/to/FreeCAD$ ./FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

user@host:/path/to/FreeCAD$
```

phaseloop, 2025-03-16:
Ok - first of all I confirmed that issue exists - workbench disappears. I'm trying to debug this, but without any success.

What is weird, from the source code (C++ one) it seems that it tries to look for custom pre/post processors in local Macro directory located at:
(if looks for files with `_pre.py` and `_post.py` suffixes

`>>> FreeCAD.getUserMacroDir()
'/home/phaseloop/.local/share/FreeCAD/Macro/'`

Also @jalapenopuzzle 

```
>>> import Path.Post
>>> print(Path.Post.__path__)  # Should show all directories being searched
['/tmp/.mount_FreeCACGdopf/usr/Mod/CAM/Path/Post']
>>> 
```

Path.Post is not available in FC 1.0 (I'm using latest development version) - I'm wrapping my head around this.

jalapenopuzzle, 2025-03-16:
> [@jalapenopuzzle](https://github.com/jalapenopuzzle) workbench issue aside - AFAIK if you plan to distribute postprocessing plugins - you should be fine by placing it in Macro directory.

Thank you very much for this suggestion.
I have confirmed that placing my post-processing script in `/home/user/.local/share/FreeCAD/Macro/` causes it to show up in the CAM -> Job Edit -> Output -> Processor drop-down menu!

This answer resolves the issue for me.

There remains the issue that the presence of `~/.local/share/FreeCAD/Mod/CAM` directory preventing the CAM workbench from loading properly, but this issue no longer blocks me from working on my post-processing plugin.

I suspect that everything discussed in this ticket is cross-platform (I can't see how any of this is specific to any particular platform).

phaseloop, 2025-03-17:
I think so - I tested it using regular Ubuntu packages, not Conda. Hasn't tested on Windows.

jalapenopuzzle, 2025-03-17:
One question remains before closing the issue. 
Is the proper way to install CAM post processing files documented in the CAM workbench documentation.

I didn't find documentation for this but I could have missed it.

LarryWoestman, 2025-03-18:
https://wiki.freecad.org/CAM_Postprocessor_Customization under "Naming convention".

jalapenopuzzle, 2025-03-18:
> https://wiki.freecad.org/CAM_Postprocessor_Customization under "Naming convention".

Thank you for finding that. If I had found it and read it, it would have saved me from making mistakes and raising this ticket. Now I'm feeling sheepish...

closed 2025-03-18 by jalapenopuzzle