FreeCAD/FreeCAD #20230: FreeCad 1.10 dev Not updating gears workbench or installing new add-ons

wyzarddoc72, 2025-03-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Attempted to update gears add-on and install Curves add-on both failed numerous times even though update was shown in the install box 

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40722 (Git) Snap 1340
Build date: 2025/03/14 04:47:53
Build type: Release
Branch: main
Hash: 9a6a98f824436656cae242ae26b043f78411ebba
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/95.958/1
Installed mods: 
  * freecad.gears.backup1742134438.4056594 (Disabled)
  * MeshRemodel.backup1742092980.7996242 (Disabled)
  * MeshRemodel 1.10.37
  * CurvedShapes.backup1742092942.0192122 (Disabled)
  * CurvedShapes.backup1742092905.6047451 (Disabled)
  * btl 0.9.9
  * Lithophane
  * MeshRemodel.backup1742093025.2806156 (Disabled)
  * freecad.gears.backup1742134476.634312 (Disabled)
  * CurvedShapes
  * A2plus 0.4.68
  * Curves 0.6.59
  * freecad.gears.backup1742093250.485393 (Disabled)
  * freecad.gears.backup1742093112.7173903 (Disabled)
  * freecad.gears.backup1737691293.1224313 (Disabled)
  * FeedsAndSpeeds 0.6.0
  * freecad.gears.backup1742093822.7837868 (Disabled)
  * freecad.gears
  * freecad.gears.backup1742092960.7291756 (Disabled)
  * freecad.gears.backup1742093186.147524 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wyzarddoc72, 2025-03-16:
unable to uninstall from snap install on ubuntu

maxwxyz, 2025-03-16:
Does it work in 1.0?

wyzarddoc72, 2025-03-16:
Update. --> problem seems to be with the Linux Snap package. Installed 1.0 from FreeCad website and works fine.

closed 2025-03-16 by wyzarddoc72