FreeCAD/FreeCAD #20231: PartDesign: Can't pocket a Part DatumPlane

Roy-043, 2025-03-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

You can create a pocket based on the old PartDesign Plane. But this is not possible with the new Part DatumPlane. Since the latter is supposed to replace the former this is something that should be addressed.

Test:
1. Open the attached file (based on https://wiki.freecad.org/Basic_Attachment_Tutorial).
2. Activate the Body.
3. Select the 'DatumPlane' (old PartDesign Plane).
4. Click pocket, and specify 'Through all' and 'Reversed'.
5. Result: OK.
6. Delete the created pocket and try to recreate it with 'DatumPlane001' (new Part DatumPlane).
7. Result: Nothing happens after invoking the pocket tool.

[attachment_tutorial.FCStd.zip](https://github.com/user-attachments/files/19272602/attachment_tutorial.FCStd.zip) (remove the .zip extension)

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/86.4394
Installed mods:
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-16:
@FreeCAD/cad-working-group is this a valid feature? IMO a plane has no size limit (infinite) and thus should not be able to create geometry itself by extruding it. It can be used to cut objects (which would be limited if it only has an arbitrary size).
The PD datum is still available in the menu.

Roy-043, 2025-03-16:
I do not see the PD datums in the menu. AFAICT they are no longer exposed in the GUI, but they are still supported, so you might create them with a macro for example.

Note that a PartDesign Plane is also infinite.

IMO this should again be labeled as a regression.

pierreporte, 2025-03-16:
I agree that the proposed test should fail. It doesn’t make sense to pad a plane. I would even say that the old behavior was a bug.

maxwxyz, 2025-03-17:
Thanks, closing it then.

Roy-043, 2025-04-05:
This has probably been closed prematurely.
https://forum.freecad.org/viewtopic.php?t=96091

Roy-043, 2025-04-05:
> It doesn’t make sense to pad a plane

This is about pocketing, not padding.

luzpaz, 2025-06-18:
The ticket doesn't where this is a regression from ? Do we know this ?

maxwxyz, 2025-06-18:
@luzpaz from 1.0

maxwxyz, 2025-06-27:
@PaddleStroke could you take a look at this? I thought there is a migration for older files or all broken?

FlachyJoe, 2025-06-27:
> it is not possible to pad or pocket an infinite plane.

If an infinite pad has no physical sens, in the contrary it should be great to allow to pocket in order to trim a solid.
I think OCC is able to use infinite shapes.

kaiwas, 2025-12-09:
> as workaround

There are 125 workarounds for every problem. But there's almost never a direct solution. It's so annoying. The "Has Workaround" label is clearly redundant.

If it's not a blocker, it's at least a significant inconvenience.
I encounter new surprises every day. All it takes is to start doing something new.

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```