FreeCAD/FreeCAD #20233: BIM_Extrude default extrusion value

semhustej, 2025-03-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature**

Priority: low

When BIM_extrude is started, the default extrude Length value is set to 10mm. In AEC applications, this is very small value. If a new user applies this without changing, he/she may not see the extrusion in the 3D view (because it is very small) and be confused.

Proposed solution:
Change the default extrusion Length value to 1000mm

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-03-17:
> It seems the value is not even remembered. It starts with 10 mm every time.

I would like to add that the default values ​​for arrays are also terribly annoying. They have to be changed in 99% of cases because most often an array is required along one axis.
And a step of 100 also seems small (new values ​​are not remembered)

[image]

I hope the situation will change someday.

semhustej, 2025-03-17:
> 
> It seems the value is not even remembered. It starts with 10 mm every time.

I can confirm this behavior. It would be nice to remember the last used settings, but that is more that this issue is originally intended for. 

If the OP is not easy to implement I would not bother with it.

furgo16, 2025-03-18:
@yorikvanhavre I had thought of the above as well, but I thought we'd then have to have a preference setting for Part Design and another one for BIM. That is, two parameters.

> After that, we can do several things inside the BIM code and modify that value or remember what the user entered.

In other words, if we would have a single parameter, say `DefaultExtrusionLength`, how would it be used in BIM? If we're using the `Part_Extrude` command, it still cannot be passed parameters.

yorikvanhavre, 2025-03-18:
Indeed. Fundamentally I think if we "wrap" the tool then we should deal with that in BIM... The Part tool itself should stay pristine.

So I would maybe do something like this, inside the BIM tool:

1. Save the value of `Mod/Part/DefaultExtrusionLength`
2. Get the value of our own prop, for ex `Mod/BIM/DefaultExtrusionLength`
3. Set `Mod/Part/DefaultExtrusionLength` to our BIM value
4. Launch the Part Extrusion tool
5. Get the value of the `ExtrusionLength` property of the last created object (maybe check that the last object created is a Part Extrude, in case the command was cancelled)
6. Save that value to `Mod/BIM/DefaultExtrusionLength`
7. Reset `Mod/Part/DefaultExtrusionLength` to the value we saved at step 1.

furgo16, 2025-03-18:
That makes sense. However, how do we deal with any issues happening after step 3 that make it impossible to restore the original `Mod/Part/DefaultExtrusionLength` value from step 1?

E.g. a FreeCAD crash, or an issue in the BIM workbench that prevents the `BIM_Extrude` command to reach step 7

yorikvanhavre, 2025-03-18:
Indeed we don't :/

We can mitigate, by placing the call to the part command inside a try statement. But there is indeed a risk. However, 1) it's not a huge risk, and 2) The user can reset the pref value manually, so it's also user-fixable...