FreeCAD/FreeCAD #20234: BIM: Default architectural project 3D view zoom settings

semhustej, 2025-03-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature**
Priority: low

When opening default BIM/Architecture default project from the Start page, the default zoom of the project is set so close to the origin, when user creates typical size object of AEC filed, such as 5m long and 2.5m high wall, the user has to then zoom out a lot to see the object. The default zoom is set up to work with objects sized ca. 200x200mm.

**Proposed solution**
If possible, set up the default zoom setting for the default Architectural project to see space ca. 20x20m or similar.
In my opinion, this would be more welcoming for new users who try the software for the first time.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-03-16:
There is a preference for the initial zoom in a document: "Camera zoom".
https://wiki.freecad.org/Preferences_Editor#Navigation

kaiwas, 2025-03-17:
> There is a preference for the initial zoom in a new document: "Camera zoom".

Thanks for the advice. I set the zoom value to 5000 mm. 

And then we need to create a part with the size 100x100x100 mm. )))

semhustej, 2025-03-17:
> I have been having this problem a lot, what I was thinking was to have some kind of heuristic to fit the camera/view to the size of the initial object added to the document. Your solution about changing the default view also sounds interesting.
> 
> Could we know beforehand when creating a new document if its a BIM document?

I am specifically talking about opening new file from the start page. I have seen that different new files open this way can have different default camera settings (for instance Draft project has different than Architectural project)

[image]

furgo16, 2025-03-17:
Interesting, one learns something new every day on FreeCAD :-)

> I have seen that different new files open this way can have different default camera settings (for instance Draft project has different than Architectural project)

Which value does the zoom have when you create a new Draft project from the button in the Start page? For me all of the buttons produce new views with the same zoom level (100 mm), but maybe I'm not interpreting the zoom value correctly (is the zoom level the `y` coordinate value on the indicator at the bottom right hand side corner in the status bar?

semhustej, 2025-03-17:
> Which value does the zoom have when you create a new Draft project from the button in the Start page? For me all of the buttons produce new views with the same zoom level (100 mm), but maybe I'm not [...]

The difference between Draft and Architectural project is in the camera orientation, not the zoom level.

furgo16, 2025-03-18:
Ok, got it now, thanks.

Looking at the code, the way the "new file" buttons work on the start page is by simply switching to the relevant workbench and issuing a set of existing commands that are relevant for the preparation of the file.

For instance when clicking on the "Parametric Part" button, a new file is created, the Part Design workbench is activated, and a new Body is added to the document. For "Assembly", it's similar, but the Assembly workbench is activated and an Assembly is added to the document. For "2D Draft", the Draft workbench is activated, and the view is set to Top. Here's an extract of the commands used to create a new Assembly and Draft document in the code: 

https://github.com/FreeCAD/FreeCAD/blob/f3bb974855fb45041b0309b3a29ecac56a2b0a8f/src/Mod/Start/Gui/StartView.cpp#L369-L384

The "BIM/Architecture" button only activates the BIM workbench.

Is there a scriptable command we could add there to change the camera zoom level?

If not, would a workaround such as creating a 10 m x 10m cube, issuing the `Std_ViewFitAll` command to effectively change the camera's zoom level and deleting the cube be acceptable? Or is it too much of a hack?

furgo16, 2025-03-18:
> Is there a scriptable command we could add there to change the camera zoom level?

Ah, I think there is. This code could be run for new BIM documents created from the start page:

```python
Gui.activeView().viewDefaultOrientation("Isometric", 20000.0)
```

However, the commands issued upon creation of new documents are called with `Gui::Application::Instance->commandManager().runCommandByName`. That is, FreeCAD named commands. The above is Python code, which I don't know if it can be executed with `commandManager()` in C++.

Any suggestions from someone more experienced in the FreeCAD code?

yorikvanhavre, 2025-03-19:
Thanks for addressing the issue @furgo16 

I'm not sure if a default Site/Building/Level structure should be added, or not... I guess some users might like it, other not. Maybe that could be controlled by a preference setting?

furgo16, 2025-03-19:
> I'm not sure if a default Site/Building/Level structure should be added, or not... I guess some users might like it, other not. Maybe that could be controlled by a preference setting?

I don't feel too strongly about it. I think it would be useful to have a default Site/Building/Level structure to:

- Enforce good design practices by providing a default structure. 
- Show new users that there is a convention for project structure. Seeing it in action will familiarize them with it. I'm thinking of myself as a new BIM workbench user a while back. The lack of a suggested convention was what I struggled at the beginning. I downloaded all BIM examples I could find to see if there was a common approach.
- If most projects contain at least a Site/Building/Level structure, I feel the exceptional ones that don't should probably not set the criteria.

Then again, I'm not an architect. It'd be good to get the experts to weigh in.

kaiwas, 2025-03-19:
If we are talking about presets, I can say that I always disable the start window. In any program (they are too heavy for constant perception). I prefer a clean interface.

That is why I start every new project with this button. Which has no presets.

[image]
It turns out that any of your decisions discussed here will not affect me )))

For example, the same button in blender

[image]

kaiwas, 2025-03-19:
> Then again, I'm not an architect. It'd be good to get the experts to weigh in.

While there is a problem with wrapping and nesting, I think it will only cause unnecessary complications.
Imagine that I have created projects where I already have a minimum set of objects in the tree. Site-Building-level.
Then I create a wall and it is created outside the level. I need to drag it manually.

This is annoying. The first thought is to delete the unnecessary and work with the walls.

That is, it is not enough to simply add some objects by default - you also need to think about how they will behave during subsequent actions. (For example, automatically set the active group).

https://github.com/user-attachments/assets/d6a5bc54-c05c-4e83-990f-9d0823feefa8

furgo16, 2025-03-19:
Thanks! I think that the objects being created outside the container is a separate issue, but I do see your point.

I was thinking about it yesterday. Perhaps the simplest approach would be to create the objects inside a (BIM) container if the container is selected in the document tree.

closed 2025-03-25 by yorikvanhavre