FreeCAD/FreeCAD #20238: Menus appear in wrong place when using two monitors with vertical offset

jalapenopuzzle, 2025-03-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am using Ubuntu 24.04 with two monitors. I have a vertical offset configured for one monitor.
The left monitor is configured higher than the right monitor. (The left monitor has a stand, and the right monitor is my laptop screen.)
The right monitor is the primary display.

- When FreeCAD is placed on the left monitor, and a menu is activated, the menu is displayed vertically disconnected from the menu bar (the menu appears way below the menu bar). The vertical disconnection distance appears similar to the left monitor raised height above the right (primary) monitor.

- When FreeCAD is placed on the right (primary) monitor, and a menu is activated, the menu is displayed connected to the menu bar as expected.

I have only tested this on Ubuntu 24.04. This issue has a high chance of being cross-platform and also needs to be tested elsewhere.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * freecad.gears 1.3.0
  * ThreadProfile 1.93.0
  * fasteners 0.5.37
  * btl 0.9.9
  * 3D_Printing_Tools
  * sheetmetal 0.7.21
```

### Subproject(s) affected?

None

### Anything else?

The menu is displayed way below the menu bar that it was expected to be attached to when FreeCAD is on the left screen:
[image]

Here is my screen configuration:
[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

karliss, 2025-03-16:
Can you check if the problem still happens when you run with `QT_QPA_PLATFORM=xcb FreeCAD`.

Early wayland support in Qt5 and first few  Qt6 versions  was quite bad and such positioning bugs were  common.  Only in later versions of QT6 the wayland support became somewhat usable. Also some distros are using patched QT5 which also somwhat fix a portion of those issues.

jalapenopuzzle, 2025-03-16:
> QT_QPA_PLATFORM=xcb

I confirm that this fixes the problem.

Regarding Qt6: It looks like Qt6.4 is packaged with Ubuntu 24.04 [1], but I haven't seen FreeCAD build instructions against qt6 for Ubuntu. I don't know if Qt6.4 counts as "early" regarding wayland defects. I don't know which Qt the Linux conda package includes.

The FreeCAD 1.0.0 that I built from source (following the Ubuntu build instructions which reference Qt5) has menus which behave correctly.

[1] https://packages.ubuntu.com/search?suite=noble&searchon=names&keywords=Qt6

karliss, 2025-03-17:
The biggest blocker for Qt6 build on Ubuntu using system packages is probably pyside. Ubuntu 24.10 is the first version which makes pyside6 available.

FEA-eng, 2025-10-01:
@jalapenopuzzle Is it still relevant or can the issue be closed?

xSetech, 2026-11-03:
The 1.0.0 AppImage could only run under X11, so I don't think this is a Wayland issue. If the reporter used the official release at the time they saw the problem, they could not have run FreeCAD under Wayland.

xSetech, 2026-11-04:
I'm able to reproduce this using X11 but not Wayland, on both the Qt from 24.04 and 26.04. This is a bug in Qt on X11.

@FEA-eng Please remove the Wayland label. I'll be back with more specifics, since it doesn't seem fixed on the latest Ubuntu LTS.

karliss, 2026-11-04:
@xSetech If you are observing problem on X11 then that's a different menu positioning bug. Same symptoms but potentially different cause.

Not sure what to say about setup used by original reporter. If the specific appimage wasn't using wayland forcing a specific backend with environment variable wouldn't had any effect.

There really were a lot of menu positioning problems in various software and UI frameworks (requiring fixes on both sides) due to wayland removing ability to arbitrarily positing windows (and popouts) and having stricter requirements for popout parenting.