FreeCAD/FreeCAD #20239: Boolean Cut causes Crash To Desktop

JDMils, 2025-03-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Checkout the forum post here:
https://forum.freecad.org/viewtopic.php?p=816901#p816901

I even recreated the model from the MangoJelly YouTube channel and it still crashed FC. This has been tested on a few releases and started around release 40317.

In essence, doing a Boolean Cut of a cylinder from a Sketch_On_Curve causes the crash. I have to use release 39980 or below in order to accomplish the task as these releases do not crash.

Sample model is included in the forum post. I've noticed a lot of instability on the releases after 39560.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39980 (Git) Conda
Build type: Release
Branch: main
Hash: 836ad91f3ddd3ec63b84b8e3f87a767d5465261d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * Curves 0.6.58
  * fasteners 0.5.36
  * Gridfinity 0.8.0
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-03-17:
This was fixed on March 11th 2025 https://github.com/FreeCAD/FreeCAD/commit/c9e88a31f28e33e7906d706982b6fe2524c06d51 and some 700 commits after the build reported here, please update to a build with at least git revision 40700 to ensure you have the fix included, I'm aware there's been problems with the Windows weekly builds but that's a separate issue.

JDMils, 2025-03-17:
Is there such a version? The only WIndows build I can see is "[FreeCAD_weekly-builds-40504-conda-Windows-x86_64-py311.7z](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-40504-conda-Windows-x86_64-py311.7z)

maxwxyz, 2025-03-19:
weekly builds are currently broken. I hope they will be up soon.

closed 2025-03-19 by maxwxyz