FreeCAD/FreeCAD #20244: BIM. Windows appearance in vers 1.0

DCMTP74, 2025-03-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

BIM. Windows apparence in plan view does not work in version 1.0 even when symbol elevation et symbol plan are set true. That worked OK in vers 0.2.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.61
  * ArchTextures
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-17:
Please provide an example file and test if this also happens for you in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

kaiwas, 2025-03-17:
I confirm. The symbol is not showing up in the latest weekly build (40655).

[image]

But I suggest waiting for the windows and doors refactoring that Yorik is currently working on.
https://github.com/FreeCAD/FreeCAD/pull/19912

maxwxyz, 2025-03-17:
@yorikvanhavre FYI

semhustej, 2025-03-17:
I am not able to reproduce the issue. In FC 1.0 and FC 1.1 I can see both symbols in plan and elevation views.

Can sample file be provided to make sure the window type in question is not 'Fixed'? This one does not show opening symbols because there is no pane which would swing.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

kaiwas, 2025-03-17:
oh! that's a nuance. would never have thought of that. probably some pop-up window with a warning is missing, stot "this type of window cannot be opened"

closed 2025-03-17 by maxwxyz

DCMTP74, 2025-03-17:
I have checked again, comparing the properties with the same file as with version 0.21. properties are identical but un version 2 the symbol plan does not show.
In this version right clik does not show the changing of hinge position.
Where do you check that the window is not "fixed".
Cannot attach FCStd File.

kaiwas, 2025-03-17:
> Where do you check that the window is not "fixed".

Preset at window creation time 

[image]

> I'll close this as it works as expected.

You are probably about 50% right.
A good practice is to "hide" or make inactive properties that do not correspond to the object. But probably for this the system design should have been thought out from the beginning.
There are a lot of such "holes" in the program at the moment.

semhustej, 2025-03-17:
@DCMTP74 

> I have checked again, comparing the properties with the same file as with version 0.21. properties are identical but un version 2 the symbol plan does not show. In this version right clik does not [...]

You have to zip the file before uploading. But it seems your window is fixed, since it does not offer you the option to switch hinge position or opening direction.

You can check by editing the window -> and than editing the glass part of the window. There is an `Opening mode` dropdown, your window probably has none set here.

[image]

DCMTP74, 2025-03-17:
Great. Thanks to semhustej and kaiwas for their time and insight