FreeCAD/FreeCAD #20245: [Compiling] Regression caused by PR17779 cannot compile with the same CMake command as last week

Syres916, 2025-03-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Specifically https://github.com/FreeCAD/FreeCAD/commit/c532683060284053439ea4cf39cd0768241c0640 is the cause of the regression. Using:

`cmake -DOCCT_CMAKE_FALLBACK=1 -DCMAKE_BUILD_TYPE=Release -DFREECAD_USE_OCC_VARIANT="Official Version" -DENABLE_DEVELOPER_TESTS=Off ../freecad-daily-source`

there is no `FREECAD_USE_PYBIND11` in the command and yet it fails with:

```
CMake Error at cMake/FreeCAD_Helpers/SetupPybind11.cmake:6 (find_package):
  By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "pybind11",
  but CMake did not find one.

  Could not find a package configuration file provided by "pybind11" with any
  of the following names:

    pybind11Config.cmake
    pybind11-config.cmake

  Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
  "pybind11_DIR" to a directory containing one of the above files.  If
  "pybind11" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:83 (SetupPybind11)
```

### Full version info

```shell
These are the libraries that are picked up when compiling is working:

OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, IfcOpenShell 0.8.1.post1, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-03-17:
Applying a `git revert c532683060284053439ea4cf39cd0768241c0640` at least gets me going.

Syres916, 2025-03-17:
Spoke too soon, part way through the compiling CAM, there's an error:

benj5378, 2025-03-17:
I don't think this is a regression. It's changing dependicies: install pybind11

benj5378, 2025-03-17:
@Syres916 the wiki page for compiling is grossly out of date by years. You are correct, that wiki/developers handbook needs to be updated with pybind11.

See discussion on #17779 for the changed dependencies.

maxwxyz, 2025-03-19:
So is this an issue in the code or does the documentation need to be updated?

benj5378, 2025-03-19:
@maxwxyz it's documentation. Everyone else had the same issue, and had to install Pybind11 package.

edi271, 2025-05-10:
> I don't think this is a regression. It's changing dependicies: install pybind11

Please show us how to install **pybind11** correctly so that **make** can find it. See https://forum.freecad.org/viewtopic.php?t=96848

FEA-eng, 2025-10-04:
@Syres916 Is this issue still relevant?

edi271, 2025-10-06:
At the moment, the problem no longer occurs. I can compile without errors.