FreeCAD/FreeCAD #20246: CMake version

feyounger, 2025-03-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

docker build--https://wiki.freecad.org/Compile_on_Docker

### Full version info

```shell
CMake Error at CMakeLists.txt:2 (cmake_minimum_required):
  CMake 3.22.0 or higher is required.  You are running version 3.16.3

-- Configuring incomplete, errors occurred!
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-17:
@adrianinsaval FYI

feyounger, 2025-03-17:
> [@adrianinsaval](https://github.com/adrianinsaval) FYI

May I ask if building with docker has not been updated for a long time?

I need to call the api in the container, is there any other good way.

adrianinsaval, 2025-03-18:
The wiki instructions are very outdated, the latest iteration of dockerfiles are in the main repo but might be outdated too

feyounger, 2025-03-18:
> The wiki instructions are very outdated, the latest iteration of dockerfiles are in the main repo but might be outdated too

Are you planning to update

luzpaz, 2025-11-25:
@feyounger ping