FreeCAD/FreeCAD #20251: Segmentation fault when opening a file and recomputing

dekay, 2025-03-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I downloaded a Freecad model of a "Microswitch mounting clip for coin ejector" [from thingiverse](https://www.thingiverse.com/thing:6829881/files). A zip file of the model is attached.

[Microswitch_Halterung_für_Cash-Return_Button.zip](https://github.com/user-attachments/files/19289331/Microswitch_Halterung_fur_Cash-Return_Button.zip)

It will segfault Freecad when I perform the following steps with the unzipped file

- open freecad from the command line
- File -> Open and load the model
- Say Yes to recompute.

Segfault info is below. Strangely enough, if I open the file directly from by KDE Desktop (right click -> Open in Freecad), I will not be asked for a recompute. It will display the model and I can open any of the sketches. However, Freecad will crash as soon as I close the sketch.

```
[dk@ryzen ~]$ freecad
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3dcd0) [0x765b4084bcd0]
#1  0x765aa777867b in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, opencascade::handle<Geom_Surface> const&, TopLoc_Location const&, double&, double&, bool*) from /usr/lib/libTKBRep.so.7.8+0x5b
#2  0x765aa7778883 in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, TopoDS_Face const&, double&, double&, bool*) from /usr/lib/libTKBRep.so.7.8+0xa3
#3  0x765aa5535b99 in ChFi3d_Builder::PerformIntersectionAtEnd(int) from /usr/lib/libTKFillet.so.7.8+0x3459
#4  0x765aa54d8dda in ChFi3d_Builder::PerformFilletOnVertex(int) from /usr/lib/libTKFillet.so.7.8+0x43a
#5  0x765aa54ddce3 in ChFi3d_Builder::Compute() from /usr/lib/libTKFillet.so.7.8+0xf63
#6  0x765aa564db9e in BRepFilletAPI_MakeFillet::Build(Message_ProgressRange const&) from /usr/lib/libTKFillet.so.7.8+0x2e
#7  0x765aa75be240 in BRepBuilderAPI_MakeShape::Shape() from /usr/lib/libTKTopAlgo.so.7.8+0x50
#8  0x765aa7d9732d in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /usr/lib/freecad/lib/Part.so+0x15d
#9  0x765aa7d97447 in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, Part::TopoShape const&, char const*) from /usr/lib/freecad/lib/Part.so+0x87
#10  0x765aa7d99299 in Part::TopoShape::makeElementFillet(Part::TopoShape const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, double, double, char const*) from /usr/lib/freecad/lib/Part.so+0x129
#11  0x765a9538be06 in PartDesign::Fillet::execute() from /usr/lib/freecad/lib/_PartDesign.so+0x286
#12  0x765b42901622 in App::DocumentObject::recompute() from /usr/lib/freecad/lib/libFreeCADApp.so+0xc2
#13  0x765aa7afca11 in Part::Feature::recompute() from /usr/lib/freecad/lib/Part.so+0x11
#14  0x765b42881444 in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x374
#15  0x765b428866ad in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /usr/lib/freecad/lib/libFreeCADApp.so+0x67d
#16  0x765b433a27d8 in Gui::Application::checkForRecomputes() from /usr/lib/freecad/lib/libFreeCADGui.so+0x418
#17  0x765b433a6c75 in Gui::Application::open(char const*, char const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x525
#18  0x765b4352f209 in StdCmdOpen::activated(int) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd79
#19  0x765b4351d47e in Gui::Command::_invoke(int, bool) from /usr/lib/freecad/lib/libFreeCADGui.so+0x26e
#20  0x765b4351d987 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad/lib/libFreeCADGui.so+0x127
#21  /usr/lib/libQt6Core.so.6(+0x1b1a49) [0x765b403b1a49]
#22  0x765b40f785a4 in QAction::activate(QAction::ActionEvent) from /usr/lib/libQt6Gui.so.6+0x154
#23  /usr/lib/libQt6Widgets.so.6(+0x2e43f9) [0x765b416e43f9]
#24  /usr/lib/libQt6Widgets.so.6(+0x2e67d3) [0x765b416e67d3]
#25  0x765b41551676 in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0xc76
#26  0x765b414ff0ca in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8a
#27  0x765b41500f4a in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x109a
#28  0x765b434acdd9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#29  0x765b40355b00 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x160
#30  0x765b414f6776 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt6Widgets.so.6+0x216
#31  /usr/lib/libQt6Widgets.so.6(+0x16a5e1) [0x765b4156a5e1]
#32  /usr/lib/libQt6Widgets.so.6(+0x16b570) [0x765b4156b570]
#33  0x765b414ff0ca in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8a
#34  0x765b434acdd9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#35  0x765b40355b00 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x160
#36  0x765b40b865cb in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt6Gui.so.6+0x3db
#37  0x765b40c01f44 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x304
#38  /usr/lib/libQt6Gui.so.6(+0x661d74) [0x765b41061d74]
#39  /usr/lib/libglib-2.0.so.0(+0x5d919) [0x765b3ead4919]
#40  /usr/lib/libglib-2.0.so.0(+0xc05d7) [0x765b3eb375d7]
#41  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x765b3ead3e15]
#42  0x765b405c575d in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#43  0x765b403606a6 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x1a6
#44  0x765b403591d6 in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0xa6
#45  0x765b433b42d4 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0xbd4
#46  freecad(+0x7de4) [0x55edacabade4]
#47  /usr/lib/libc.so.6(+0x27488) [0x765b40835488]
#48  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x765b4083554c]
#49  freecad(+0x81e5) [0x55edacabb1e5]
```

### Full version info

```shell
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-21:
does it work in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

dekay, 2025-03-21:
Well, I'll be darned. It works fine with FreeCAD_weekly-builds-40655-conda-Linux-x86_64-py311.AppImage.

When opened in V1.0.0, a sketch and a pad associated with that sketch were not aligned over top of each other. When opened in the dev version, that doesn't happen anymore. Better yet, I did a Save As... in the Dev version and can now work with it in the V1.0.0 version. Awesome! Thank you @maxwxyz!

Should this bug be closed or should it stay open in case the root cause is known and it can be backported to the stable version?