FreeCAD/FreeCAD #20254: FreeCAD-1.0.0-39109 quits unexpectedly requesting printing a TechDraw page

seb1953, 2025-03-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

FreeCAD-1.0.0-39109 quits unexpectedly requesting printing a TechDraw page.
OS: MacOs Sonoma 14.5

### Full version info

```shell
OS: macOS 14.5
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * freecad.gears 1.3.0
  * woodworking 0.21.2.33771.0.24.1
  * fasteners 0.5.37
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-03-17:
I cannot reproduce.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

```ruby
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

seb1953, 2025-03-18:
Replicating the problem, it happens when requesting printing through the **print icon** in techdraw wb. If printinting is requested through the FC menu (File->Print), it works fine.

seb1953, 2025-03-19:
Yes, button TechDraw_PrintAll.

Tamagotono, 2025-04-23:
I have been fighting this issue as well.
I am using the weekly appimages, so not sure if I should be starting a new issue or not.  One point that differs is that Freecad crashes for me when attempting to "Print" to pdf as well as exporting to pdf.  Verified this is true when activated from the /TechDraw_PrintAll menu option as well.  It crashes even when trying to do a print preview.

I did some segmenting and the last working appimage that worked was: "FreeCAD_weekly-builds-41232-conda-Linux-x86_64-py311.AppImage". 

The 3 appimages that have been released since (41264, 41314, 41354) all segfault.

Looks like the breakage happened between 4/11 and 4/14 (the two build dates), so hopefully will not be too difficult to track down.

Information from the last appimage that **worked**:
```
OS: NixOS 24.11 (Vicuna) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41232 (Git) Conda AppImage
Build date: 2025/04/11 14:31:02
Build type: Release
Branch: main
Hash: 1983052e381b5e5dbc78778186dc8f85a23163e5
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/109.201/1
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * sheetmetal 0.7.22
  * kicadStepUpMod 11.6.8
  * Cables 0.1.4
  * Gridfinity 0.11.8
  * FreeGrid 2.2.0
  * Defeaturing 1.2.4
```

Information from the earliest appimage that **failed**:
```
OS: NixOS 24.11 (Vicuna) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda AppImage
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/109.201/1
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * sheetmetal 0.7.22
  * kicadStepUpMod 11.6.8
  * Cables 0.1.4
  * Gridfinity 0.11.8
  * FreeGrid 2.2.0
  * Defeaturing 1.2.4
```

Here is what got spit out to the terminal when it crashed.
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/libc.so.6(+0x40620) [0x7fb058a66620]
#1  0x7fb05a0aabf0 in QGraphicsItemPrivate::setVisibleHelper(bool, bool, bool, bool) from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/../lib/libQt6Widgets.so.6+0x90
#2  0x7faf0634facb in TechDrawGui::ViewProviderTemplate::setMarkers(bool) from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/lib/TechDrawGui.so+0xab
#3  0x7faf0633d12a in TechDrawGui::ViewProviderPage::setTemplateMarkers(bool) const from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/lib/TechDrawGui.so+0x7a
#4  0x7faf062b933e in TechDrawGui::PagePrinter::renderPage(TechDrawGui::ViewProviderPage*, QPainter&, QRectF&, QRect&) from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/lib/TechDrawGui.so+0xce
#5  0x7faf062bb6a5 in TechDrawGui::PagePrinter::printPdf(TechDrawGui::ViewProviderPage*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/lib/TechDrawGui.so+0x465
#6  0x7faf062b5f42 in TechDrawGui::MDIViewPage::printPdf() from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/lib/TechDrawGui.so+0x1c2
#7  0x7fb05bdfde2f in Gui::Command::_invoke(int, bool) from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/../lib/libFreeCADGui.so+0x21f
#8  0x7fb05bdfe1df in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/../lib/libFreeCADGui.so+0x12f
#9  /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/../lib/libQt6Core.so.6(+0x30a527) [0x7fb05910a527]
#10  0x7fb05998d583 in QAction::triggered(bool) from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/../lib/libQt6Gui.so.6+0x43
#11  0x7fb059990972 in QAction::activate(QAction::ActionEvent) from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/../lib/libQt6Gui.so.6+0xf2
[... 3 lines cut ...]
#15  0x7fb05a3a40f4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/../lib/libQt6Widgets.so.6+0x84
#16  0x7fb05a3a4f8d in QApplication::notify(QObject*, QEvent*) from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/../lib/libQt6Widgets.so.6+0xddd
#17  0x7fb05bda0499 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/../lib/libFreeCADGui.so+0xc9
#18  0x7fb05914d29a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/../lib/libQt6Core.so.6+0x16a
[... cut ...]
#34  0x7fb05bcfb5b4 in Gui::Application::runApplication() from /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/../lib/libFreeCADGui.so+0x734
#35  /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/freecad(+0x5ca0) [0x563e1be1dca0]
#36  /lib/libc.so.6(+0x2a27e) [0x7fb058a5027e]
#37  /lib/libc.so.6(__libc_start_main+0x89) [0x7fb058a50339]
#38  /home/jason/.cache/appimage-run/b9e2f29a0684e1d8b80c5d610a64d107671501d6a6d3f61f4dcf2232f86f280f/usr/bin/freecad(+0x602a) [0x563e1be1e02a]
```

Tamagotono, 2025-04-26:
I just checked and the latest Appimage `./FreeCAD_weekly-builds-41416-conda-Linux-x86_64-py311.AppImage ` does not exibit this behavior. It works _almost_ perfectly.  The only issue now is that it does not append ".pdf" to the file name when exporting.  I'll file a separate bug for that. 

Just for clarity, this latest version DOES work.
`OS: NixOS 24.11 (Vicuna) (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.41416 (Git) AppImage
Build date: 2025/04/25 15:50:11
Build type: Release
Branch: main
Hash: b2a9f0f7d2cbaf633ecd42f137c3e17871a0fb68
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/109.201/1
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * sheetmetal 0.7.22
  * kicadStepUpMod 11.6.8
  * Cables 0.1.4
  * Gridfinity 0.11.8
  * FreeGrid 2.2.0
  * Defeaturing 1.2.4
`

xtemp09, 2025-04-26:
I'd like to point out that the post above does not include information about the operating system, as well as the versions of _PySide_ and _Shiboken_.

Tamagotono, 2025-04-28:
Nice catch!
I Just verified that When I go to About -->Libraries-->[Qt for Python(Shibokin / PySide) 
under Version, it is blank. So there is a reporting issue there.  If there is another way of getting this information, let me know how and I'll be happy to supply it.

Syres916, 2025-05-01:
I've just tested the latest 1.0 branch (what will become version 1.0.1) and it works as expected, could the original issue be a Mac specific issue?

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39275 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: deb68996f357bf02161b449036042d5e39ff0867
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-Ribbon 1.8.1.2 (Disabled)
  * btl 0.9.9
  * FreeCAD-themes 2025.1.7
  * Silk 0.1.6
  * sheetmetal 0.7.22
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.30.3
  * ChangeAppFont
  * Quetzal 1.4.4
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Curves 0.6.61
  * QetWireManager
  * A2plus 0.4.68
  * ThreadProfile 1.94.0
```

luzpaz, 2025-05-10:
> I'd like to point out that the post above does not include information about the operating system, as well as the versions of PySide and Shiboken.

@xtemp09 this was a separate issue fixed in  
* #21213

luzpaz, 2025-05-10:
> I've just tested the latest 1.0 branch (what will become version 1.0.1) and it works as expected, could the original issue be a Mac specific issue?

@johnocfii could you check to see if this is macOS related ?

JohnOCFII, 2025-05-11:
> [@JohnOCFII](https://github.com/JohnOCFII) could you check to see if this is macOS related ?

I'm not seeing an issue on macOS.  I did not see an attached FCstd file, so I just whipped up a basic example.  Export to PDF from the button worked fine for me.

```
OS: macOS 15.4.1
Architecture: arm64
Version: 1.1.0dev.41632 (Git)
Build date: 2025/05/09 19:27:21
Build type: Unknown
Branch: main
Hash: ce368cb7a0dd6a3f57e5cabc3e7745c5669683a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2024.9.1
  * AddonManager 2025.5.1a
  * Telemetry 1.0.1
```

luzpaz, 2025-05-11:
Thanks. Closing as this seems fixed in 1.1dev

closed 2025-05-11 by luzpaz