FreeCAD/FreeCAD #20257: CAM Workbench - Array for operation doesn't set coolant flag

RWL-69, 2025-03-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using FreeCAD CAM workbench, create an operation that requires coolant.  Duplicate the operation using the Array tool, create the G-Code and the coolant flag is not enabled in the array operations.  Seen this in three different versions of FreeCAD and on both Linux and Windows.  I tested with the weekly build.

Missing coolant will cause damage to tools or/and material.

### Full version info

```shell
OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/85.8968/1
Installed mods: 
  * DesignSPHysics 0.7.1(26-11-2024)
  * Assembly4 0.50.18
  * Plot 2024.11.26
  * Curves 0.6.59
  * symbols_library
  * ose-workbench-core-0.1.0a4
  * CurvedShapes.backup1713545489.8854659 1.0.5 (Disabled)
  * Defeaturing 1.2.3
  * FeedsAndSpeeds 0.6.0
  * FEM_FrontISTR 0.2.0
  * slic3r-tools
  * fasteners 0.5.37
  * OSE3dPrinter 0.1.0
  * InventorLoader 1.5.1
  * ose-workbench-core-0.1.0a4.zip
  * MeshRemodel 1.10.37
  * toSketch 1.0.1
  * freecad.gears 1.3.0
  * LCInterlocking
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.21
  * Manipulator 1.5.7
  * kicadStepUpMod 11.5.5

AND

OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Classic/Qt default
Installed mods: 
  * DesignSPHysics 0.7.1(26-11-2024)
  * Assembly4 0.50.18
  * Plot 2024.11.26
  * Curves 0.6.59
  * symbols_library
  * ose-workbench-core-0.1.0a4
  * CurvedShapes.backup1713545489.8854659 1.0.5 (Disabled)
  * Defeaturing 1.2.3
  * FeedsAndSpeeds 0.6.0
  * FEM_FrontISTR 0.2.0
  * slic3r-tools
  * fasteners 0.5.37
  * OSE3dPrinter 0.1.0
  * InventorLoader 1.5.1
  * ose-workbench-core-0.1.0a4.zip
  * MeshRemodel 1.10.37
  * toSketch 1.0.1
  * freecad.gears 1.3.0
  * LCInterlocking
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.21
  * Manipulator 1.5.7
  * kicadStepUpMod 11.5.5
```

### Subproject(s) affected?

None

### Anything else?

A simple CAM job that pointed out the missing M7 command in the Array part of the code.  (Coolant On:Mist)
M7 is only at the beginning of the source operations.  (Coolant Off:Mist) M9 is turned off at the end of those operations. 

[CAM_Bug.zip](https://github.com/user-attachments/files/19299255/CAM_Bug.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

phaseloop, 2025-03-18:
Issue confirmed. It seems in array we copy path movements but do not copy any of the operation setup g-code
```
  for b in base:
                    pl = FreeCAD.Placement()
                    pl.move(pos)
                    np = Path.Path(
                        [cm.transform(pl) for cm in PathUtils.getPathWithPlacement(b).Commands]
                    )
                    output += np.toGCode()

```

Same bug is present in operation -> SimpleCopy

phaseloop, 2025-03-18:
Edit: I found the root cause. Array operation can have multiple bases but `def determine_coolant_mode(obj)` expects object to have one base and therefore does not detect if flooding should be enabled. 

We have similar issue if two bases use different tool controllers - so we use controller from first base (and emit warning).

sliptonic, 2025-03-18:
I'm not a fan of the array operation.  IMHO, it should only accept a single base and should probably be re-implemented as a dressup.  

In its present form, it's going to cause all kinds of problems like this and it will get worse if the user tries to use it in conjunction with Work Coordinate System offsets.

phaseloop, 2025-03-18:
AFAIK base being a list is deeply baked into CAM - so we can at least return some error. Also - are we fine with breaking change like this?

Maybe Array instead of common Path.Path() for everything at once should return an array of some intermediate objects (toolController, coolantFlag, etc, Path.Path()) and then it job exporter later should handle it.

sliptonic, 2025-03-18:
Base as a list is deeply baked in as a target for an operation and the bases should be solids (or 2D objects).  That's by design and key to operations working on multiple solids at once.  

In this case, the operation (array) is targeting other operations not solids.  That's a mutant and, I think, a mistake.  

A dressup is meant to target an operation and that's why it can only target one.   The relationship to tool controllers is key. 

I think we should make this change.  An array that targets multiple operations with different tool controllers will almost definitely break things with multiple WCS depending on how the user configures ordering.

phaseloop, 2025-03-18:
I agree. But in the meantime still there is a long legacy of Array tool. Quick fix would be to only allow bases with the same coolant mode and expose the mode for postprocessors.

sliptonic, 2025-03-18:
We don't know how many people are using the Array with multiple bases.  I would suggest:
1) a forum topic to find out.
2) deprecate the ability to have multiple bases as early as possible.
3) If a LOT of people have a LOT of models using arrays of multiple operations, write a conversion script to split them into multiple arrays.

phaseloop, 2025-03-18:
IMHO there is no point in depreciating having multiple bases - once we have array dressup we should be encouraging people to use the dressup and finally remove the original operation from FC one day.

The only edge case to handle now is checking if all bases use same coolant setting - this is broken anyway so we are not going to break it more - only preventing people from exporting wrong g-code.

RWL-69, 2025-03-18:
Until this issue if properly fixed, could a pop-up warning be generated to the user indicating that there may be an issue with coolant?  At least that will prevent, what happened to us, from damaging the work piece and tool.

Or have the post processing code look at the original process to set the coolant flag.  It would be better if the process of the array was part of the original process instead of being its own.

phaseloop, 2025-03-18:
AFAIK it's easier to fix - under the assumption that we do not allow having multiple bases with different coolant setting (even if we break some people CAM jobs, at least we will prevent them from damaging tools).

Moving array to a dressup operation is a different thing we should do soon.

sliptonic, 2025-03-18:
I think we should make the effort to fix it right.  If you just add a check that the bases have the same coolant setting, you have to decide what to do if they don't.  Warn the user?  Prevent creation?  Take the first/last setting?  Something else?

The coolant setting is actually trivial.  The fact that the array allows multiple bases creates a much more concerning condition if the base operation have different tool controllers.  This will probably cause the order-by to break.

If we just prevent the array from having multiple bases, both of those problems go away and you basically have a dressup.  So why not migrate the array operation logic to a dressup?  It should be very straight-forward and it's the right thing to do.

phaseloop, 2025-03-18:
I see some issues raised about an array dressup (no idea if valid): https://github.com/FreeCAD/FreeCAD/issues/20275

I'm not saying we should not migrate this to a dressup. What I'm saying is:

- Array dressup will be a new option in a `Path Dressup` menu in a separate code file (right?)
- To not break existing files/jobs - old Array will need to live there too for some time, probably with an edit popup that user should migrate to a dressup instead.
- While legacy Array operation is still present, and we wait for users to migrate - we can at least make it less error prone.

What I mean by making it less error prone:
- if array uses multiple bases, check if all bases use same coolant option. If not - raise an error and prevent g-code from being generated
- if array uses multiple bases which use different tool controllers - raise an error and prevent g-code from being generated.

RWL-69, 2025-03-18:
I don't understand the comments about multiple bases but assuming that it may mean different parts.

There should be no tool change in the array and the array should be tied to the process it is based on so it follows the original process, so dressup is the best path forward.

sliptonic, 2025-03-18:
> I'm not saying we should not migrate this to a dressup. What I'm saying is:
> 
> * Array dressup will be a new option in a `Path Dressup` menu in a separate code file (right?)

Let's not overthink it.  The whole array.py file is less than 500 lines and most of that is routine stuff. The actual getPath method is about 100 lines of simple python.  

Yes, it's a breaking change but it's minimal impact on the user.  Worst case scenario, they have to recreate the array by dressing up the operation.  What we should do is make sure users stop using the old array as soon as possible.

Dressups and operations are quite similar.  I would suggest creating the new dressup, and copying the core logic of array over to it.   

Remove the array operation from the menus so users cannot create new ones but old ones still work.

Add an 'on document restored' check for the array to display a warning message and notice that it will be deprecated. 

> I don't understand the comments about multiple bases but assuming that it may mean different parts.
> 
> There should be no tool change in the array and the array should be tied to the process it is based on so it follows the original process, so dressup is the best path forward.

Operations and dressups have a base property that refers to another document object where it can get data.  For all operations (other than array) the base is a list shapes to work on.

Dressups don't work on shapes.  Instead, they take the path from another operation and modify it.  Holding tags, lead in/out, dogbones, etc are all dressups.  

Array _should_ be a dressup.  It shouldn't be working on the base shape to generate a toolpath.  Instead, it should be modifying an existing path by making one or more copies of it.  Dressups only allow a SINGLE base.  Whoever implemented array implemented it as an operation which means it can have multiple bases but the bases are other operations.  There in lies the problem.

phaseloop, 2025-03-19:
I implemented first version of a dressup. It seems when it comes to a tool controller and coolant there is a wild-wild west because every nested dressup misses coolant settings too because obj.Base points to another dressup and not to a operation having `CoolantMode` property :D

So probably I need to refactor dressups to have a common base class implementing that.

sliptonic, 2025-03-19:
There's a utility function in Path/Dressup/Utils.py that returns the baseOp for dressup.  It will recurse to the original op.  So it should be easy to find the coolant setting.

RWL-69, 2025-03-20:
Would it be better to tie the processes to the tools used?  If the tool is set to use coolant, then it is applied across all processes that use that tool.  I am new to this so I may not understand many aspects of the process.

I also feel that processes need to be ordered to minimize tool changes.  I attribute that to the user to order the processes in the correct order.  That is how I arranged my last job.  I don't know if it is a good idea to have the post processor do that which is a different subject.