FreeCAD/FreeCAD #20260: Slider Joint in Assembly Workbench doesn't respect workspace units

dogunter, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If I create a Slider Joint between two parts in the native Assembly workbench, and then try to set "Length Max" and "Length Min", the values are always in millimeters no matter what units I have been working in (Imperial Decimal in my case).

### Full version info

```shell
OS: macOS 15.3.2
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * woodworking 0.21.2.33771.0.24.1
  * ExplodedAssembly
```

### Subproject(s) affected?

Assembly

### Anything else?

This happens whether I'm working in Safe Mode or not. My defaults is set to "Imperial Decimal" for new documents.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

drwho495, 2025-03-18:
try in 1.1dev please

dogunter, 2025-03-18:
> try in 1.1dev please

I just did, and now when I even attempt to set the "Enable Length Max" property to "true" for the slider properties, the program immediately crashes. The same happens if I attempt to set the "Enable Length Min" property to "true". 

Thus, there is no way for me to even see what units are being used now.

Here's my version info:
OS: macOS Sequoia (15.3)
Architecture: arm64
Version: 1.1.0dev.40843 (Git) Conda
Build date: 2025/03/18 01:59:29
Build type: Release
Branch: slider_units_issue
Hash: f3bb974855fb45041b0309b3a29ecac56a2b0a8f
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/81/1
Installed mods: 
  * woodworking 0.21.2.33771.0.24.1
  * ExplodedAssembly

drwho495, 2025-03-18:
> > try in 1.1dev please
> 
> I just did, and now when I even attempt to set the "Enable Length Max" property to "true" for the slider properties, the program immediately crashes. The same happens if I attempt to set the "Enable Length Min" property to "true".
> 
> Thus, there is no way for me to even see what units are being used now.
> 
> Here's my version info: OS: macOS Sequoia (15.3) Architecture: arm64 Version: 1.1.0dev.40843 (Git) Conda Build date: 2025/03/18 01:59:29 Build type: Release Branch: slider_units_issue Hash: [f3bb974](https://github.com/FreeCAD/FreeCAD/commit/f3bb974855fb45041b0309b3a29ecac56a2b0a8f) Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1 Locale: English/United States (en_US) Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/ Logical DPI/Physical DPI/Pixel Ratio: 72/81/1 Installed mods:
> 
> * woodworking 0.21.2.33771.0.24.1
> * ExplodedAssembly

could be another mac build bug

dogunter, 2025-03-18:
> could be another mac build bug

I suspect as much. I just created #20279 with all the info.

dogunter, 2025-03-25:
Now that #20279 has been resolved, I have tested this issue in Dev 1.1 build 40971 and the problem still exists. Slider joint min/max length units are locked to millimeters independent of units chosen for the document.

FEA-eng, 2025-10-02:
Seems fixed:

[image]

@PaddleStroke Can you confirm the issue should be closed?

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

dogunter, 2025-10-02:
Using the latest release (1.0.2) as well as the latest weekly build (2025.10.01) I can set the lengths in the appropriate units when creating the joint, in the Tasks menu. It is when I highlight the joint and look at the properties that I see all the settings are in metric units. Must be the internal way things are done.

[image]

PaddleStroke, 2025-10-03:
Joint object properties are all App::PropertyFloat when they should be App::PropertyLength or App::PropertyAngle.

PaddleStroke, 2025-10-03:
@yorikvanhavre I am having trouble changing the type of the property. I found in bim this example : 

```
            type = obj.getTypeIdOfProperty('ArchSketchEdges')
            if type == "App::PropertyIntegerList":
                oldIntValue = obj.ArchSketchEdges
                newStrValue = [str(x) for x in oldIntValue]
                obj.removeProperty("ArchSketchEdges")
                obj.addProperty("App::PropertyStringList","ArchSketchEdges","Structure",QT_TRANSLATE_NOOP("App::Property","Selected edges (or group of edges) of the base ArchSketch, to use in creating the shape of this BIM Structure (instead of using all the Base shape's edges by default).  Input are index numbers of edges or groups."), locked=True)
                obj.ArchSketchEdges = newStrValue
```

But when I try to do the same : 

```
        if hasattr(joint, "Distance2") and joint.getTypeIdOfProperty("Distance2") == "App::PropertyFloat":
            old_value = joint.Distance2
            joint.removeProperty("Distance2")
            joint.addProperty(
                "App::PropertyLength",
                "Distance2",
                "Joint",
                QT_TRANSLATE_NOOP(
                    "App::Property",
                    "This is the second distance of the joint. It is used only by the gear joint to store the second radius.",
                ),
                locked=True,
            )
            joint.Distance2 = old_value
```

I get this error 
`<class 'NameError'>: Property AssemblyDebug#Joint.Distance2 already exists`
It looks like the removeProperty is not working. Or not fast enough. But the BIM code is doing the same thing apparantly

closed 2025-10-06 by chennes