FreeCAD/FreeCAD #20261: Deleting a simple LSC crashes freecad.

leoheck, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Deleting a simple LSC crashes freecad. This kills me as a user. It should not crash.

Todo: Add steps to reproduce it and a sample file.

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x705e95e45330]
#1  0x705e976294d8 in QTreeViewPrivate::viewIndex(QModelIndex const&) const from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5+0x48
#2  0x705e9762cea4 in QTreeView::visualRect(QModelIndex const&) const from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5+0xa4
#3  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5(+0x3bf75c) [0x705e975bf75c]
#4  0x705e975cb686 in QAbstractItemView::viewportEvent(QEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5+0x346
#5  0x705e964b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Core.so.5+0x7b
#6  0x705e9737039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#7  0x705e9917e2f9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0xc9
#8  0x705e964b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Core.so.5+0x108
#9  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5(+0x1759e6) [0x705e973759e6]
#10  0x705e973766b4 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5+0x474
#11  0x705e97376e7c in QApplicationPrivate::sendSyntheticEnterLeave(QWidget*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5+0x26c
#12  0x705e973aa690 in QWidgetPrivate::hideChildren(bool) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5+0x140
#13  0x705e973aa807 in QWidgetPrivate::hide_helper() from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5+0x117
#14  0x705e973ae9a8 in QWidgetPrivate::setVisible(bool) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5+0x368
#15  0x705e9941a52f in Gui::TreeWidget::onUpdateStatus() from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0x7ff
#16  0x705e9941bd10 in Gui::TreeWidget::checkTopParent(App::DocumentObject*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0xf0
#17  0x705e99773f51 in Gui::SelectionSingleton::checkSelection(char const*, char const*, char const*, Gui::ResolveMode, Gui::SelectionSingleton::_SelObj&, std::__cxx11::list<Gui::SelectionSingleton::_SelObj, std::allocator<Gui::SelectionSingleton::_SelObj> > const*) const from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0x4c1
#18  0x705e99774f9d in Gui::SelectionSingleton::isSelected(char const*, char const*, char const*, Gui::ResolveMode) const from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0x11d
#19  0x705e997757ab in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0x10b
#20  0x705e9977d897 in Gui::SelectionSingleton::slotDeletedObject(App::DocumentObject const&) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0x3d7
#21  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADApp.so(+0x24151f) [0x705e9804151f]
#22  0x705e98265490 in App::Application::slotDeletedObject(App::DocumentObject const&) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADApp.so+0x10
#23  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADApp.so(+0x24151f) [0x705e9804151f]
#24  0x705e9805f38d in App::Document::removeObject(char const*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADApp.so+0xdd
#25  0x705e980dfb67 in App::DocumentPy::removeObject(_object*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADApp.so+0x77
#26  0x705e980dfe04 in App::DocumentPy::staticCallback_removeObject(_object*, _object*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADApp.so+0x24
#27  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x705e98806068]
#28  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x705e987e5b5c]
#29  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x705e987f1532]
#30  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x705e988abd1d]
#31  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x705e988ab794]
#32  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x705e988cabf3]
#33  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x705e988c669a]
#34  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x705e988ba9c9]
#35  0x705e97b4ceac in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADBase.so+0x6c
#36  0x705e991cf504 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0x94
#37  0x705e991ef944 in StdCmdDelete::activated(int) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0x7b4
#38  0x705e991d961f in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0x23f
#39  0x705e991d99df in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0x12f
#40  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x705e964eb6f2]
#41  0x705e9736a462 in QAction::triggered(bool) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5+0x32
#42  0x705e9736ce2b in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5+0xab
#43  0x705e99239e40 in Gui::ShortcutManager::onTimer() from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0x100
#44  0x705e9923ae2d in Gui::ShortcutManager::checkShortcut(QObject*, QKeySequence const&) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0x52d
#45  0x705e9923d136 in Gui::ShortcutManager::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0x96
#46  0x705e964b67ca in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Core.so.5+0x6a
#47  0x705e973703e8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Widgets.so.5+0xb8
#48  0x705e9917e2f9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libFreeCADGui.so+0xc9
#49  0x705e964b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Core.so.5+0x108
#50  0x705e96b62d63 in QShortcutMap::dispatchEvent(QKeyEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Gui.so.5+0x423
#51  0x705e96b6368b in QShortcutMap::tryShortcut(QKeyEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Gui.so.5+0x5b
#52  0x705e96b0e60a in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Gui.so.5+0x16a
#53  0x705e96b3129f in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Gui.so.5+0x7f
#54  /tmp/.mount_FreeCADhPmFG/usr/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so(+0xbf1b) [0x705e90ab8f1b]
#55  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x705e964eb6f2]
#56  0x705e90c1eb5f in QDBusPendingCallWatcher::finished(QDBusPendingCallWatcher*) from /tmp/.mount_FreeCADhPmFG/usr/plugins/platforms/../../lib/libQt5DBus.so.5+0x2f
#57  0x705e964e0830 in QObject::event(QEvent*) from /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/libQt5Core.so.5+0x260
[... 7 lines cut ...]
#65  /tmp/.mount_FreeCADhPmFG/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x705e94b140ce]
[... cut ...]
#70  /tmp/.mount_FreeCADhPmFG/usr/bin/freecad(+0x7906) [0x5b049e9af906]
#71  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x705e95e2a1ca]
#72  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x705e95e2a28b]
#73  /tmp/.mount_FreeCADhPmFG/usr/bin/freecad(+0x7c8e) [0x5b049e9afc8e]
```

### Full version info

```shell
FreeCAD 1.1.0, Libs: 1.1.0devR40655 (Git)
FreeCAD_weekly-builds-40655-conda-Linux-x86_64-py311.AppImage
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-03-18:
I've just deleted _Local Coordinate System_ (LCS). It does not crash on _Windows 10_ and _KUbuntu 24_. This issue lacks certain steps to reproduce and an example file.

luzpaz, 2025-03-18:
Full About info is necessary for everyone involved

xtemp09, 2025-04-02:
Is this issue still relevant?

closed 2025-04-12 by leoheck