FreeCAD/FreeCAD #20263: FEM: Pipeline functions issues

maxwxyz, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

There are some issues when inserting functions in the result pipeline:
1. The default scale of a plane is not matched to the document size. It would be great if it had a constant size on the display, like the transform dragger or the new datum features, where they are displayed the same size independent of the zoom level.
2. The function has an icon of an unknown object and transparency cannot be toggled.
3. When deleting the function object, individual objects are still left behind but placed at root level, outside the analysis container

Demo:

https://github.com/user-attachments/assets/03c51cef-9dec-4422-894e-a2c857e5bac9

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/133.772
Installed mods: 
  * Cables 0.1.4
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.59
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Gridfinity 0.11.2
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.21
  * Silk 0.1.5
  * Telemetry 1.0.0
  * Telemetry.backup1742214704.5089447 1.0.0 (Disabled)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-18:
@FEA-eng FYI

closed 2025-04-29 by chennes

FEA-eng, 2025-05-15:
Point 1 is actually #14523 so this issue may remain closed.