FreeCAD/FreeCAD #20264: SVG: Straight lines are imported as B-spline

Pendelf, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Straight lines are imported as B-spline if you translate the geometry into a sketch

https://github.com/user-attachments/assets/02395171-ad62-46d0-a7cb-53c66422540b

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * AirPlaneDesign 0.4.1
  * btl 0.9.9
  * CurvedShapes 1.0.13
  * Curves 0.6.59
  * fasteners 0.5.37
  * PieMenu 1.11.2
  * Road 2025.2.22
  * sheetmetal 0.7.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-03-18:
Probably the sketcher needs some way to convert B-splines into simpler geometry. At least straight sections.

phaseloop, 2025-03-18:
Can this be caused by "too good" tolerance when even straight SVG line is being treated as curve?

Something @aprospero tries to change

aprospero, 2025-03-28:
> Can this be caused by "too good" tolerance when even straight SVG line is being treated as curve?
> 
> Something [@aprospero](https://github.com/aprospero) tries to change?

The svg import code doesn't change simple edge types into more complex ones (the other way around may sometimes happen). So I suggest, that the splines are already in the svg. Inkscape does some weird things when it comes to creating paths from objects. Lately I found a path that contained beziers with relative poles and endpoints all zero...

aprospero, 2025-04-08:
I have to correct myself: due to OCCT limitations when  transforming shapes, straight lines indeed often degenerate to b-splines in the process of svg parsing.
Shoogens once added an alternative way to apply transformations, where at least orthogonal transformations didn't result in this kind of deformation. Most Transformations are orthogonal (scale/translate/rotate) so this alternative would have been the solution to this problem. But unfortunately the alternative was removed due to a regression if I understand right:
[https://tracker.freecad.org/view.php?id=2062](https://tracker.freecad.org/view.php?id=2062)