FreeCAD/FreeCAD #20272: Close Python editor (by menu or by X icon) crashes

yomgui1, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**This crash happens with a fresh build under pixi, but not with a weekly build.**

1. Start freecad
2. Start a new macro (that open the python editor with a blank script)
3. Close the editor by menu or by the sub-window icon.The application ends with this line in the console:

```
ASSERT failure in Gui::EditorView: "Called object is not of the correct type (class destructor may have already run)", file /storage/projects/freecad-src/.pixi/envs/default/include/qt6/QtCore/qobjectdefs_impl.h, line 129
```

_Also tested under --safe-mode (crash occures)._

### Full version info

```shell
OS: Fedora Linux 40 (Workstation Edition) (i3/i3/xcb)
Architecture: x86_64
Version: 1.1.0dev.40722 (Git) Conda
Build date: 2025/03/14 04:47:53
Build type: Release
Branch: main
Hash: 9a6a98f824436656cae242ae26b043f78411ebba
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/102.406/1
Installed mods: 
  * Render 2024.12.15
  * FeedsAndSpeeds 0.6.0
  * cadquery_module 2.1.0
  * InventorLoader 1.5.1
  * sheetmetal 0.7.21
  * fasteners 0.5.37
  * BillOfMaterials 1.0.2
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-21:
Does this also happen in 1.0 and only in your local builds?

xtemp09, 2025-03-26:
@yomgui1, you compiled _FreeCAD_ against a debug version of _Qt 6_. This version adds additional assert, so when a signal is sent to a class when its destructor is run, _FreeCAD_, like any other _Qt_ program, crashes.

More information you can find here: https://github.com/FreeCAD/FreeCAD/issues/15325#issuecomment-2225402451

commit 9d9ee7d9 (2025-03-26): Disconnect signals in destructor to avoid the assert

Closes #20272.
closed 2025-03-26 by hyarion