FreeCAD/FreeCAD #20273: Python console tooltip does not display method arguments upon completion

furgo16, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using the autocompletion feature, two types of tooltips are shown:

1. When hovering over the list of members offered for autocompletion
   [image]
2. When having autocompleted e.g. a method with the <kbd>Tab</kbd> keyboard key

In the second case, the first line of the docstring is used. Generally, that docstring line will not contain the arguments, as it would be repeating the argument description in the subsequent lines. Thus the argument names will not be shown in the tooltip.

Instead of using the first line of the docstring, it would be great if the arguments could be extracted (by parsing the full docstring, or from type hinting info if there's any, or by any other method), and then shown in the tooltip.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40722 (Git) Snap 1340
Build date: 2025/03/14 04:47:53
Build type: Release
Branch: main
Hash: 9a6a98f824436656cae242ae26b043f78411ebba
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

### Subproject(s) affected?

None

### Anything else?

What I've noticed is that some docstrings in the code explicitly mention the argument names in the first line, as a workaround. That leads to duplication, and another place to change (and to potentially be forgotten) whenever there are API changes.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xelathan, 2025-04-01:
> it would be great if the arguments could be extracted (by parsing the full docstring, or from type hinting info if there's any, or by any other method), and then shown in the tooltip.

If the method does not take any arguments, then what do we expect to be displayed in the tooltip dialog?

furgo16, 2025-04-02:
If we can take inspiration from existing IDEs, vscode shows the type hinting info on top of the tooltip, and the docstring at the bottom.

For a method with arguments:
[image]

For a method without arguments:
[image]

xelathan, 2025-04-03:
> some docstrings in the code explicitly mention the argument names and the method name in the first line, as a workaround

This seems to be the main issue. The docstrings do not have a unified format. The preexisting code expected all docstrings to have the method name and args in the first line.

In the example you used above for `Draft.getCloneBase()`, this not only caused the tooltip to display incorrect info, but also caused the cursor to be placed incorrectly; its supposed to be placed in between `()` because `getCloneBase` has arguments and it supposed to allow the user to type them in easier.

I'll put up a PR soon addressing this by making these independent of the docstrings.

xelathan, 2025-04-03:
Also it would be nice if we could refactor the docstrings, removing all the method name and args in the first line.

[image]

> vscode shows the type hinting info on top of the tooltip

It would also be nice if we could add type labels and return labels on top of the Python APIs as it would improve the tool tip here.

These can probably be addressed in separate refactoring issues.

furgo16, 2025-05-08:
@maxwxyz could you link https://github.com/FreeCAD/FreeCAD/pull/20630 to this issue? Thanks!