FreeCAD/FreeCAD #20277: Assembly: TNP when features in part intersect

drwho495, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Before you close this issue, I do not believe this is the same as: https://github.com/freecad/FreeCAD/issues/19363

This is more of a subsect of that issue. Also I know how to fix this; I just need a good way to implement the fix

Anyways, what is happening is when a pocket in a body intersects with a fillet in the body, it causes TNP in assembly. This is not a program-wide issue because you can setup a LCS on the affected edge, and it throws a missing reference exception as needed.

Now for the fix, if you reference the tip of the part in Reference 1 and Reference 2 of the joint, then the issue is fixed. Is there a reason for not including the tip in those two params as is?

https://github.com/user-attachments/assets/0787d0fb-dd6c-4491-97fa-5061d433bccd

[Asm_TNP_PD.zip](https://github.com/user-attachments/files/19324431/Asm_TNP_PD.zip)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/161.377/1
Installed mods: 
  * Render 2024.12.15 (Disabled)
  * ArchTextures
  * Assembly3 0.12.2
  * CadbaseLibrary 2.0.0
  * Behave-Dark-Colors 0.1.1
  * SearchBar 1.0.2
  * WebTools 1.0.0
  * Manipulator 1.5.7
  * POV-Ray-Rendering
  * cadquery_module 2.1.0
  * Ondsel-Lens 2024.11.29.01 (Disabled)
  * freecad.gears 1.3.0
  * ExplodedAssembly
  * fasteners 0.5.37
  * cadquery_module.backup1741030188.0603402 (Disabled)
  * FreeCAD-Telemetry 1.0.0beta
  * QuickMeasure 2022.10.28
  * FreeCAD-Ribbon 1.8.0
  * assembly2
  * Freecad-Built-in-themes-beta 1.2.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Assembly

### Anything else?

Here is the code I made to fix the issue, it works but could be done without creating the references in the first place

```
def addTipToReference(self, joint, refIndex):
        reference = None

        if refIndex == 0:
            reference = joint.Reference1
        elif refIndex == 1:
            reference = joint.Reference2
        
        referenceName = reference[1][0]
        refNameSplit = referenceName.split(".")
        
        if len(refNameSplit) != 3:
            refObj = App.ActiveDocument.getObject(refNameSplit[0])
            if refObj is not None:
                if hasattr(refObj, "Tip") and refObj.Tip is not None:
                    tip = refObj.Tip
                    newRefName = refNameSplit[0] + "." + tip.Name + "." + refNameSplit[1]
                    reference[1][0] = newRefName
                    reference[1][1] = newRefName

                    if refIndex == 0:
                        joint.Reference1 = reference
                    elif refIndex == 1:
                        joint.Reference2 = reference
```

in onChanged()
```
if hasattr(joint, "Reference1") and joint.Reference1 is not None:
        self.addTipToReference(joint, 0)
        
if hasattr(joint, "Reference2") and joint.Reference2 is not None:
        self.addTipToReference(joint, 1)
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

drwho495, 2025-03-18:
@PaddleStroke

drwho495, 2025-03-18:
This fix seems to break auto-changes, i'll come up with a fix

drwho495, 2025-03-18:
So I made a rather poor fix by creating two new properties called Reference1Tip and Reference2Tip (it was the only way to see if the edge was missing in the tip of the model, im still trying to find a better way to do this). Then I checked to see if the edge was missing in that property and changed missingReference accordingly.

PaddleStroke, 2025-03-19:
Currently in assembly we do not store the name of the feature. ie the reference stores :

`assembly.body.face1` 
instead of 
`assembly.body.pad2.face1`

Then assembly always consider the tip feature of the body.

Which was a design we took in the early days of the workbench. I don't remember precisely the pros and cons of both approach. Changing this now would be a significant undertaking.

So perhaps we can fix this without changing that

drwho495, 2025-03-19:
> Currently in assembly we do not store the name of the feature. ie the reference stores :
>
> `assembly.body.face1` instead of `assembly.body.pad2.face1`
>
> Then assembly always consider the tip feature [...]

As stated earlier, if you create two more properties with the tip reference then that is updated properly, then you can setup some logic to error the joint if the tip reference goes missing.

drwho495, 2025-07-08:
https://github.com/FreeCAD/FreeCAD/pull/17768 seems to fix