FreeCAD/FreeCAD #20279: Setting "Enable Length Max" or "Enable Length Min" for Slider Joint immediately crashes 1.1.0dev

dogunter, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Create two simple parts. Create a a native Assembly with a simple Slider Joint between the two parts. Try to set the Slider Joint properties "Enable Length Max" or "Enable Length Min" to "true" and the program immediately crashes.

I attached a file that demonstrates this and a gzipped crash report output.

### Full version info

```shell
OS: macOS Sequoia (15.3)
Architecture: arm64
Version: 1.1.0dev.40843 (Git) Conda
Build date: 2025/03/18 01:59:29
Build type: Release
Branch: slider_units_issue
Hash: f3bb974855fb45041b0309b3a29ecac56a2b0a8f
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/81/1
Installed mods: 
  * woodworking 0.21.2.33771.0.24.1
  * ExplodedAssembly
```

### Subproject(s) affected?

None

### Anything else?

[slider_unit_issue_dev1.1.FCStd.gz](https://github.com/user-attachments/files/19326891/slider_unit_issue_dev1.1.FCStd.gz)

[crash_report.txt.gz](https://github.com/user-attachments/files/19326906/crash_report.txt.gz)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

drwho495, 2025-03-18:
this isn't occuring on the latest linux daily

maxwxyz, 2025-03-19:
please test if newer weeklies will be available

dogunter, 2025-03-19:
> please test if newer weeklies will be available

As soon as working weeklies become available for MacOS again I will.

dogunter, 2025-03-25:
This is not longer an issue as of the working MacOS Arm build 40971. Problem fixed.

closed 2025-03-25 by dogunter