FreeCAD/FreeCAD #20280: Draft: Draft_Move with a negative fractional imperial value doesn't work

psommerfeld, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As illustrated in the attached video:
1. Select object in Draft or BIM WB.
2. Run Draft_Move
3. In either Local delta X, Y, or Z field, enter some negative imperial amount, with both feet and inches, for instance: -4'7"
4. Press Tab
5. You will notice the value you entered decreased by some amount. Also notice the "Length" field doesn't match your entered vaue as it should.
6. Shift-tab to go back to that field. Press Tab again
7. The value decreases again. This can be done repeatedly

### Full version info

```shell
OS: macOS 14.0
Architecture: arm64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/127.5
Installed mods: 
  * fasteners.backup1686222008.1392 0.3.50 (Disabled)
  * ProDarkThemePreferencePack 1.0.0
  * BIM021 2021.12.0
```

### Subproject(s) affected?

Draft

### Anything else?

Positive X'Y" values work.
Negative -X' works.
Only negative with feet and inches seems to fail, ie -X'Y"

https://github.com/user-attachments/assets/cf75091a-4912-4873-a184-ab79517745ba

[copy bug.FCStd.zip](https://github.com/user-attachments/files/19327164/copy.bug.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-03-19:
This is indeed a know problem, when using a negative sign inside a feet + inch expression, the result is not properly computed. IIRC there was an issue open for that already, with a long discussion, but I can't find it back...

Roy-043, 2025-03-19:
-X'Y" is interpreted as -X' + Y"

Workaround: enter -X'-Y".

psommerfeld, 2025-03-19:
Thanks for the workaround idea. Sure needs to be fixed though!

> Workaround: enter -X'-Y".

Roy-043, 2025-04-05:
Related to: #11345.

psommerfeld, 2025-11-06:
It's still an issue