FreeCAD/FreeCAD #20287: BIM: Project Setup dialog window width is too small

furgo16, 2025-03-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The dialog window that appears when clicking on the `BIM_ProjectManager` command in the toolbar needs to be wider and not crop widgets horizontally, which also makes a horizontal scroll bar to appear:

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40846 (Git) Snap 1344
Build date: 2025/03/19 00:35:33
Build type: Release
Branch: main
Hash: 382720b82e4c6c81f7e1cdd3a33272047d7a3642
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-03-19:
It looks like this is again OS dependent, this is how the dialogue looks like on Windows:

[image]

```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

yorikvanhavre, 2025-03-19:
I wonder if there isn't a way to "automatically resize to contents"... QtDesigner does it so it must exist :)

Here the problem might also be that the contents are inside a QFrame, which makes it inherently resizable... Maybe we should remove the QFrame?

furgo16, 2025-03-19:
> I wonder if there isn't a way to "automatically resize to contents"... QtDesigner does it so it must exist :)

I'm no Qt expert, but I've been wondering this all the time. Is this not one of the essential functions of Qt's widgets?

I was surprised that the recent fixes to window widths involved hardcoding the width value or even modifying stylesheets.

semhustej, 2025-03-19:
It seems to me under GNOME and non-classic style the text is much larger than other DE. Maybe someone with KDE could check.

semhustej, 2026-05-02:
I tested also with GNOME desktop and I am not able to reproduce the issue.

[image]

@furgo16 it seems to me you are the only one who is able to reproduce the issue. Can you chime in with your opinion if it's worth for to pursue this issue?

@SAAIANIRUDH-S have you been able to reproduce the issue in safe mode?

```
OS: Debian GNU/Linux 13 (trixie) (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.2.0dev.20260429 (Git shallow) AppImage
Build date: 2026/04/29 07:07:16
Build type: Release
Branch: grafted,grafted
Hash: a785d40ac4c37f559b0b82fbaf9bf7dc68a468e4
Python 3.11.14, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/158.196/1

```