FreeCAD/FreeCAD #20288: BIM: saving Project Setup presets results in error

furgo16, 2025-03-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Start FreeCAD, switch to BIM workbench
2. Click on the `BIM_ProjectManager` command button in the toolbar
3. On the "Project setup" window, click on the `Save preset...` button
4. Expected: the preset is saved
5. Actual: Python stacktrace

```python
Traceback (most recent call last):
  File "/snap/freecad/1344/usr/Mod/BIM/bimcommands/BimProjectManager.py", line 438, in savePreset
    + str(int(self.form.groupNewDocument.isChecked()))
AttributeError: 'PySide2.QtWidgets.QDialog' object has no attribute 'groupNewDocument'
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40846 (Git) Snap 1344
Build date: 2025/03/19 00:35:33
Build type: Release
Branch: main
Hash: 382720b82e4c6c81f7e1cdd3a33272047d7a3642
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-03-19:
I can confirm the issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

yorikvanhavre, 2025-03-19:
Duh.. Should be groupNewProject. Fixed in https://github.com/FreeCAD/FreeCAD/pull/20295

PR #20295 (2025-03-22): BIM: Fixed project manager's save/restore
Fixes #20288

I left the original name inside the stored file, so older ones stay compatible...
commit c1614001 (2025-03-22): BIM: Fixed project manager's save/restore - fixes #20288
closed 2025-03-22 by chennes