FreeCAD/FreeCAD #20289: BIM: Project Setup cannot add human figure

furgo16, 2025-03-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Start FreeCAD, switch to BIM workbench
2. Click on the `BIM_ProjectManager` command button in the toolbar
3. On the "Project setup" window, leave the `Add a human figure` checkbox checked
4. Press the OK button
5. **Expected**: a human figure is shown in the model
6. **Actual**: an empty shape named `Human` appears on the document tree, but it seems to be empty and it is not visible in the 3D view

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40846 (Git) Snap 1344
Build date: 2025/03/19 00:35:33
Build type: Release
Branch: main
Hash: 382720b82e4c6c81f7e1cdd3a33272047d7a3642
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-03-19:
Actually, I wonder, since the grid has a human figure, does it still make sense to add one here? We could just remove that checkbox.

furgo16, 2025-03-19:
I was thinking that too. That said, that setting is really difficult to find. It's under the Draft workbench preferences, and if I remember correctly it is only shown for BIM projects (when the grid is shown).

I've always found I want to have a human figure to grasp relative sizes at a glance, but it's so confusing to set up in FreeCAD, that I never know when or if it will be shown.

It might be simpler to have a command button to add a human figure that is an actual part that can be moved around and hidden or shown when the user wants to. It would also make it possible to add more than one figure, in case one wants to have at least a figure on each level.

yorikvanhavre, 2025-03-20:
> It might be simpler to have a command button to add a human figure that is an actual part that can be moved around

This is a very good idea. I would really like to have a command to add these kinds of "cosmetic stuff" inside a BIM project. Not only human figures, but also things like trees, cars, .... These could even be parametric and offer different "styles"...

commit fc4aa06f (2025-03-20): BIM: HumanFigure.brep: fix typo in create_qrc.py

Fixes #20289

My bad. Because of this typo the brep file was not included in Arch_rc.py.
commit c4a9c75b (2025-03-20): BIM: HumanFigure.brep: fix typo in create_qrc.py

Fixes #20289

My bad. Because of this typo the brep file was not included in Arch_rc.py.
closed 2025-03-20 by yorikvanhavre