FreeCAD/FreeCAD #20290: BIM: Project Setup in native mode can add multiple projects

furgo16, 2025-03-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

## Steps to reproduce

1. Start FreeCAD, switch to BIM workbench
2. Click on the `BIM_ProjectManager` command button in the toolbar
3. On the "Project setup" window, select the "Create a Native IFC project in the current document" radio button option
4. Click on the OK button => OK, project is created 
6. Repeat steps 2 and 3
7. Expected: there is already a project in the document, no new project is added
8. Actual: a new project is added in addition to the existing one in the document

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40846 (Git) Snap 1344
Build date: 2025/03/19 00:35:33
Build type: Release
Branch: main
Hash: 382720b82e4c6c81f7e1cdd3a33272047d7a3642
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-03-19:
When working in locked native IFC mode, the project manager is actually supposed to EDIT the current project. There must be something wrong :) I'll have a look

semhustej, 2025-03-19:
> When working in locked native IFC mode, the project manager is actually supposed to EDIT the current project. There must be something wrong :) I'll have a look

When following the steps and using the "Create a Native IFC project in the **current document**" option, it does not switch to locked IFC mode. This makes sense to me from the description.

When repeating steps 2 and 3, new native IFC project is created inside of the hybrid document, which is the father. This makes sense to me again. I am not sure though, whether FreeCAD was meant to work with two separate IFC projects inside one FCstd document - seems like not. If that is the case, I guess this option should be grayed out if FreeCAD detects current document with IFC project already inside of it.

That is my understanding from the description in the interface.

FEA-eng, 2025-11-06:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251104 (Git shallow)
Build date: 2025/11/04 19:07:44
Build type: Release
Branch: main
Hash: af014a1bb44264d749ba9522917f0d1ac76c4bbf
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```