FreeCAD/FreeCAD #20296: Sketcher: SketchObject getPoint function cluttering

PaddleStroke, 2025-03-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

Before we had a single function, 64 lines of code. Very basic and straightforward. Very simple to understand. A series of if else for each case. No way this would have gone wrong. And most importantly for developers, when you were browsing SketchObject.cpp it was not cluttering.

And now it's 165 lines. And its cluttering the code.
On top of that this is using templates. So I don't think we can say it is reducing cognitive complexity.

@AjinkyaDahale please consider reverting this.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kadet1090, 2025-03-19:
@FreeCAD/code-quality-working-group FYI

hyarion, 2025-03-19:
I guess this is the commit in question: https://github.com/FreeCAD/FreeCAD/commit/5f86d7f5a870d283521b2d94cfa9db98d4a643e7

AjinkyaDahale, 2025-03-19:
In this particular case, have no opinion either way about whether to keep one function or many. However, the switch-case structure makes things clearer so IMHO that should be kept.

Additionally, and it might not be a problem at all in the pre-refactor code, but we should keep the logic for each curve type in it's own block, whether within if-else or as separate functions.

A PR is welcome, but I'd like to avoid making any back and forth changes (again, in this particular case) myself.

kadet1090, 2025-03-19:
I am highly against reverting this change. 

Now each curve type has properly separated function that is easy to parse individually without needing to fish it from function that has everything. For someone outside of sketcher development, this function was not easy to understand as everything was in the same place and thus was hard to navigate, even if it had "only" 64 lines of code. Using templates and template specification in that context is a good C++ way to solve problem of having different logic for different classes.

hyarion, 2025-03-19:
The refactoring did make it a bit more to read, but the original version wasn't easy to read either.
There must be a better solution than to revert it to the original wall of code, as that wasn't particularly easy to read either.

The new functions only seem to be used as helper functions inside the c++ file. @AjinkyaDahale is the plan to make use of it elsewhere or could they be made inline functions in the cpp file, or moved out to some other file?

benj5378, 2025-03-19:
> Now for anyone who has not majored in computer science with solid CPP knowledge, the code looks harder to understand. Templates may be good CPP. In practice random non-cpp expert but who knows python could-have-been contributors are scared by these.

Idk why you are so confused about the templates. There were no ever template parameters, it was purely function overloading, marked as template (even though it didn't need to be marked as one).

```cpp
// Actual code
template <> 
Base::Vector3d SketchObject::getPointForGeometry<>(const Part::GeomPoint *geomPoint, PointPos PosId) 
{
    return Base::Vector3d(); 
}

// If the code wasn't unneeded marked template
Base::Vector3d SketchObject::getPointForGeometry(const Part::GeomPoint *geomPoint, PointPos PosId) 
{
    return Base::Vector3d(); 
}
```
⬆️ Either the characters `template<>` obfuscates your brain, or you didn't read the code. Come on man.

> And as an actual user, as in an actual user of sketcher source code to make new features and fix bugs, which afaik you are not

> Now tell me, how often do you browse the sketchobject.cpp file? Because of
course, to be highly against it you are often doing it and actually using
getPoint function right?

> What code base users needed was getPoint2d not some useless separated
functions that are never used outside of getPoint.

This is just falling so low. Maybe it would have been faster fixing the code (#20308, not a revert) when you saw it? Took me 40 minutes...

closed 2025-03-20 by PaddleStroke

PaddleStroke, 2025-03-20:
Ok whatever I'm not loosing more time here.
But I will say this again. Who here spend times browsing this file? Not one of you (aside from Ajinkya of course, but he's the only one that admited that he may have overdone it here). So it's easy to speak in favor of the conceptual code quality and how easy it is to read this new version. 

Start spending days on sketcher's code then you will have an honest opinion on this kind of well wishing code quality refactoring. Sketcher is full of those. To the point it is painful to work on sketcher.

Was this file a long list of if/else? Yes but what bad came from it? Nothing because you never had to open this damn function to read it!!! Because it was totally transparent on what it was doing.

But of course, you know better.

pieterhijma, 2025-03-20:
Firstly, thanks everyone for discussing this! (despite some negativity, but I can understand). It shows that we are taking code quality very serious now and that is very positive. 

My 2 cents on this issue are the following: The pattern we have here is a "dispatch function", but not on only one level but on two (type of geometry, and then the point position). In the previous function the two-level dispatch logic was merged in one large function and in my humble opinion this merging of dispatching inherently creates complexity (what if there were a third level for example). So, I'm in favor of dispatching on one level and then on the second in the way is done here. @benj5378 found a way to dispatch in a different way and perhaps that is the way forward, but as @hyarion points out, it is also not without problems.

I believe @PaddleStroke actually highlights a different and in my opinion an independent problem as well, which is that the code is hard to navigate. Having separate functions for dispatching on the second level makes it even harder because more functions are added. Indeed, having more functions in a 12k lines file does not help but in my humble opinion we should mainly address the fact that SketchObject is 12k lines.

So, in my opinion pulling apart two-level dispatch logic as is done here is preferable. Perhaps we should investigate dispatching in a different way as @benj5378 is experimenting with. However, we should also not be blind to the fact that SketchObject is huge and difficult to navigate in general. The fact that this file is so large puts pressure on decisions such as pulling apart two-level dispatch logic in the way is done here (hence this discussion). It is likely that the large file size puts pressure on all kinds of other decisions to make the code more readable as well.

Any ideas for that, perhaps from @AjinkyaDahale?