FreeCAD/FreeCAD #20299: importing excel files with dragged equations causes crash

pageeddie, 2025-03-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In excel, you can drag down an equation into neighboring cells. If this is done, and the resulting excel file is imported into FreeCAD, it will crash FreeCAD.
This occurs in FreeCAD 1.0, as well as 1.1.0dev 40504.
I have attached a very simple spreadsheet that illustrates the problem - column A has manually entered equations. Column E has identical equations, but some have been created via "drag-copy". Once cell E5 is reached, FreeCAD crashes.
To replicate:
1) start FreeCAD
2) create a new document
3) import the attached file

[bug_demo.xlsx](https://github.com/user-attachments/files/19341111/bug_demo.xlsx)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/93.2987
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-03-19:
I cannot reproduce the crash with the stable and weekly bundles on _Windows 10_ and _KUbuntu 24_. Perhaps, the issue lacks certain steps.

To make luzpaz happy, I'll append the following.

```ruby
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

pageeddie, 2025-03-19:
xtemp09 - thank you for looking into this.
Perhaps "crash" is the wrong word - FreeCAD stays up.
But the spreadsheet does not properly import.
Cell E5 is where the problem starts. It is not populated, nor are cells A6 or E6.
I tried this just now in Ubuntu as well, with FreeCAD 0.2 as well as 1.0 - exact same results. As soon as a dragged down equation is hit, all subsequent importing stops.

Here is what appears in my report view:

09:55:41  process Strings: 
09:55:41  string:  Manually entered equations
09:55:41  string:  Dragged down equations
09:55:41  pyException: Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "C:\Users\eddie\AppData\Local\Programs\FreeCAD 1.0\bin\Lib\site-packages\freecad\module_io.py", line 16, in OpenInsertObject
    getattr(importerModule, importMethod)(objectPath, *importArgs, **importKwargs)
  File "C:\Users\eddie\AppData\Local\Programs\FreeCAD 1.0\Mod\Spreadsheet\importXLSX.py", line 473, in insert
    handleWorkSheet(myDom, theSheet, stringList)
  File "C:\Users\eddie\AppData\Local\Programs\FreeCAD 1.0\Mod\Spreadsheet\importXLSX.py", line 319, in handleWorkSheet
    handleCells(row.getElementsByTagName("c"), actSheet, strList)
  File "C:\Users\eddie\AppData\Local\Programs\FreeCAD 1.0\Mod\Spreadsheet\importXLSX.py", line 351, in handleCells
    actCellSheet.set(ref, fTrans.translateForm(theFormula))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\eddie\AppData\Local\Programs\FreeCAD 1.0\Mod\Spreadsheet\importXLSX.py", line 171, in translateForm
    self.getNextToken(actExpr)
  File "C:\Users\eddie\AppData\Local\Programs\FreeCAD 1.0\Mod\Spreadsheet\importXLSX.py", line 186, in getNextToken
    if len(theExpr) > 0:
       ^^^^^^^^^^^^
<class 'TypeError'>: object of type 'NoneType' has no len()

maxwxyz, 2025-03-19:
Does it work in 1.0?

pageeddie, 2025-03-19:
It does not work in 1.0. Same behavior. So just to clarify, I have tried
Windows 11, versions 1.0 and 1.1.0dev 40504
Ubuntu 24.04, version 1.0 and 0.20
All variations do the same thing - problems begin starting at cell E5

xtemp09, 2025-03-20:
@luzpaz, I strongly disagree with the title. The Pythonic script to import XLSX file, _importXLSX.py_, does not handle certain cells.

xtemp09, 2025-12-10:
@luzpaz, yes, much better.

maxwxyz, 2026-11-07:
There is a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/4c2639624e504a104d4a1a3a68316f4e9e940290

It imports shared formulas and adjusts the cell references for the dragged copies. I tried the regression test, it fails on upstream and passes with this patch.