FreeCAD/FreeCAD #20302: BIM: add tile support for slabs

furgo16, 2025-03-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the same way a wall has the MakeBlocks and associated properties, it would be useful for a slab to have an equivalent MakeTiles property. With it, the slab could be subdivided into tiles of the given size, so that they could be counted.

### Full version info

```shell
1.1.0dev
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

markkomartin, 2025-03-19:
It would be a nice feature. 
Do you think it would be possible to decide the starting point of the tiles? 
Otherwise you get wrong results

furgo16, 2025-03-19:
> Do you think it would be possible to decide the starting point of the tiles?

If it's possible, I don't know. I just opened the issue and we're still discussing. Whether it's a good idea, I think so, yes.

furgo16, 2025-03-19:
As a workaround, I've just noticed you can use a wall as the base for a slab. So you can create a horizontal wall, divide it in blocks the size of the times, select the wall and press the slab command.

yorikvanhavre, 2025-03-20:
That's a very good idea!

Indeed it is totally possible to create tiles with the wall tool. So here I think I would extract the functionality from the wall tool, and put it somewhere where both objects can use it.

Note that i'm in the middle of a wall tool refactor at #19554 where I'm already separating the blocks making functionality in an own function. I'd wait until that's finished, that will simplify the work here

Roy-043, 2025-03-20:
If you create an array of tiles (solids) and a 'full' slab you should be able to create a Part_Common from the two.

furgo16, 2025-03-21:
> If you create an array of tiles (solids) and a 'full' slab you should be able to create a Part_Common from the two.

That's also a great idea. I was actually thinking "why can't we use standard array tools to create tiles"?

However, I must be doing something wrong with [my testing](https://github.com/user-attachments/files/19385719/TestTilesFail.zip):
- The result takes ages to compute
- It looks like Space Invaders are coming to get me

[image]

Roy-043, 2025-03-21:
For the boolean operation to work there must be space between the tiles. The joints can't be zero.

furgo16, 2025-03-21:
That fixed it, thanks!

@yorikvanhavre @Roy-043 Is there a specific reason why we couldn't use this operation (array + `Part_Common`) to create the tiles instead of the [`MakeBlocks` custom code](https://github.com/FreeCAD/FreeCAD/blob/e987441542657fa5a8397d9d005e486df96c0af2/src/Mod/BIM/ArchWall.py#L410)?

Is it because it's not easily reversible if `MakeBlocks` were set to `True` and then `False`?

Roy-043, 2025-03-23:
Dividing walls into blocks is more complex. Using an array in the case of a multi-segment wall would not work I think. The code to create blocks has some limitations, but it does allow the creation of running bond patterns. Such a pattern would be harder to accomplish with an array-type solution.

furgo16, 2025-03-23:
Good point, thank you. But for floors, would it make sense to use link arrays to reduce complexity and to gain performance?

As you said, tthe limitation would be that running bonds for floor tiles would not be supported (or be more difficult to achieve). Would that not be a reasonable compromise?

Roy-043, 2025-03-24:
A link array is more efficient than a non-link array. But the boolean result derived from a link-array is not efficient any more. It would be interesting to see if it is possible to create brick and tile patterns where all whole units are links, and only the cut bricks/tiles are unique shapes. But I do not know how that should be done.

For the sake of performance attaching a Draft_Facebinder to the top of the slab and hatching that is probably better. But there are not doubt architects who would also want to model tiles.

furgo16, 2025-03-24:
Thanks, I had not considered that the `Part_Compound` result would nullify the effect of performance from the link array. I wonder if another approach could be possible: iterate through each tile of the compound and replace them with a link to a tile created with the same length x width. I'll need to see if they're iterable at all, and if there is a way to obtain their dimensions.

Regarding the hatching, another disadvantage would be losing the count of full and broken tiles. I'm sensitive to this aspect, because that's what I'm trying to do right now :)

paullee0, 2025-04-04:
Regis' Hatch Generator seems interesting approach and related?

Any idea if he is in github?

[[WIP] Parametric HATCH GENERATOR.](https://forum.freecad.org/viewtopic.php?p=807087#p807087)

[image]
[image]

furgo16, 2025-04-05:
Good call. https://github.com/chopinregis/Freecad_Stuffs_I_Share/tree/main/Parametric%20Hatch%20Generator @chopinregis FYI

paullee0, 2025-04-05:
Thanks !  How to ping in github?  @chopinregis seems not working?

furgo16, 2025-04-05:
We already pinged him both correctly. It's up to him to reply if he's got notifications activated and if/when he has some time.

chopinregis, 2025-04-07:
yes the hatch generator can create tiles. it's not perfect but it can create tiles. so you can create floor boundaries, and the tile object, and then use the hatch generator and generate the hatch, and then once you have your tile profile just apply a Slab or Panel object to it.

[image]

[image]

[image]

[image]

chopinregis, 2025-04-07:
and ofcourse you can add the more features to your base tile shape

[image]

chopinregis, 2025-04-07:
The tool is not perfect, but for now it works and can acheive some great amout of things.

furgo16, 2026-02-04:
@sonejostudios thanks for the heads up, check out https://github.com/FreeCAD/FreeCAD/pull/27222 - this is one approach, which I've been working on recently. There is a somewhat similar approach (https://forum.freecad.org/viewtopic.php?p=807087#p807087) by @chopinregis, but I don't know the status of that or whether the code is available.

It would be good to discuss whether we want to include Tessellator functionality for ArchCovering if/when it gets merged, or whether it's best to have a specialized add-on for specific functions.

chopinregis, 2026-02-04:
@furgo16 the code is available and you can view it here (https://github.com/chopinregis/Freecad_Stuffs_I_Share/tree/3445af08c3c565513f1b2d7ce5a4759b9f4d8cb8/Parametric%20Hatch%20Generator)