FreeCAD/FreeCAD #20303: Exporting to VRML (wrl) results in segmentation fault (11)

dhoepfl, 2025-03-19:
I have a reproducible crash on macOS in both FreeCAD 1.0 and 0.21:

1. Start FreeCAD
2. Create a new, empty file
3. Open Part workbench
4. Create a primitive (e.g. a sphere)
5. Select the sphere in the model tree
6. File → Export (as VRML, Unnamed-Sphere.wrl)

Other things I tried:

* Imported an STEP file; Selected the object; File → Export ⇒ Crashes.
* Exporting in the Part Workbench, using Part → Export CAD file… does not crash but does not export the object from the STEP file when exporting to STEP (if it does, it does not export the appearance/colors).
* Convert the object to a mesh, then export the mesh ⇒ Crashes.

<details><summary>Crash Report</summary>
<code>
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [23706]
Path:                  /Applications/#ThirdParty/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               602

Date/Time:             2025-03-18 12:15:58.2689 +0100
OS Version:            macOS 15.3.2 (24D81)
Report Version:        12
Bridge OS Version:     9.3 (22P3060)
Anonymous UUID:        0D8CC528-81E4-A952-F15F-D695818CD750

Sleep/Wake UUID:       2347B814-D8E4-47E1-9A37-13C1A6B09F8F

Time Awake Since Boot: 410000 seconds
Time Since Wake:       320 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [23706]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 4475195376
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10abe1000-10ac01000    [  128K] r-x/r-x SM=COW  /Applications/#ThirdParty/FreeCAD.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libCoin.80.0.3.dylib          	       0x110121560 SoCallbackAction::~SoCallbackAction() + 80
1   libCoin.80.0.3.dylib          	       0x1101423ac SoToVRML2ActionP::~SoToVRML2ActionP() + 284
2   libCoin.80.0.3.dylib          	       0x110140568 SoToVRML2Action::~SoToVRML2Action() + 40
3   libFreeCADGui.dylib           	       0x10d4128c5 Gui::SoFCDB::writeToVRML(SoNode*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>&) + 277
4   libFreeCADGui.dylib           	       0x10d4129a2 Gui::SoFCDB::writeToVRML(SoNode*, char const*, bool) + 66
5   libFreeCADGui.dylib           	       0x10d414fd0 Gui::SoFCDB::writeToFile(SoNode*, char const*, bool) + 144
6   libFreeCADGui.dylib           	       0x10d0d78f4 Gui::Application::sExport(_object*, _object*) + 1412
7   libpython3.11.dylib           	       0x10b2c8a8e cfunction_call + 94
8   libpython3.11.dylib           	       0x10b3bca7e _PyEval_EvalFrameDefault + 225566
9   libpython3.11.dylib           	       0x10b382f96 _PyEval_Vector + 4918
10  libpython3.11.dylib           	       0x10b381be9 PyEval_EvalCode + 249
11  libpython3.11.dylib           	       0x10b42d948 run_mod + 168
12  libpython3.11.dylib           	       0x10b42fe68 PyRun_StringFlags + 104
13  libFreeCADBase.dylib          	       0x10aecbb5f Base::InterpreterSingleton::runString(char const*) + 111
14  libFreeCADGui.dylib           	       0x10d1b00e6 Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) + 150
15  libFreeCADGui.dylib           	       0x10d0a5a5c Gui::Application::exportTo(char const*, char const*, char const*) + 1580
16  libFreeCADGui.dylib           	       0x10d1c392f StdCmdExport::activated(int) + 2751
17  libFreeCADGui.dylib           	       0x10d1aed81 Gui::Command::_invoke(int, bool) + 673
18  libFreeCADGui.dylib           	       0x10d1aeaa5 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 341
19  libQt5Core.5.15.15.dylib      	       0x10f6522ae 0x10f505000 + 1364654
20  libQt5Widgets.5.15.15.dylib   	       0x10c47e0d1 QAction::triggered(bool) + 55
21  ???                           	    0x7fa857027e80 ???
</code>
</details> 

I also tried it on Linux and Windows were the export works just fine.

maxwxyz, 2025-03-19:
@dhoepfl please add full about info

dhoepfl, 2025-03-20:
(Sorry for adding it to the wrong repo.)

> [@dhoepfl](https://github.com/dhoepfl) please add full about info

That one?

OS: macOS 15.3.2
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

Is there anything else I could provide?

When started from the Terminal, all it says is “Segmentation fault: 11”, --log-file also did not contain anything that seemed useful.

I also tried it in a newly added user (to rule out a problem with my setup) but it happens there, too.
(BTW: On the very first start as a new user, the menu bar didn’t change from what it is at the Splash-Screen to a full menu bar, so there was no File menu. On the second start it worked.)

Here’s what I do, recorded in my normal account:

https://github.com/user-attachments/assets/73556fd2-80da-46ab-ae79-d9b5dfb7c3b8

maxwxyz, 2025-03-20:
Yes, could you also test if it is an issue in the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

dhoepfl, 2025-03-20:
> Yes, could you also test if it is an issue in the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Already did so, that build fails to start on macOS:

[Logmessages when starting](https://github.com/user-attachments/files/19364102/startup.txt)

I also tested 0.21, also crashes.

dhoepfl, 2025-03-27:
I think I found it. See [here](https://github.com/coin3d/coin/issues/556).

FEA-eng, 2025-08-30:
@dhoepfl Is it still relevant?

dhoepfl, 2025-08-30:
> [@dhoepfl](https://github.com/dhoepfl) Is it still relevant?

Yes. I can still reproduce it with FreeCAD 1.0.2 (release version) and with the weekly build 2025-08-27 by following the steps shown in the video from my earlier comment.

I think the bug has been [fixed in Coin3D](https://github.com/coin3d/coin/issues/556) and I think it is included in Coin3D 4.0.5 but as far as I can tell, both versions of FreeCAD link version 4.0.3.

luzpaz, 2026-03-09:
Is there a PR to update Coin3D dependency ? If so, that should close this ticket.

dhoepfl, 2026-09-16:
Just tested: Still crashes in 1.1.3.

Closing bugs “to keep the backlog tidy” is ridiculous. What _additional_ information am I supposed to provide here? I analyzed the crash in depth, found it in Coin3D, pointed them to the problem, it got fixed there.

I will not open a pull request to update the dependency without testing it by building FreeCAD locally first. Setting up a build environment for FreeCAD is something I am not currently able to do.

That being said, I have no idea why the release version still uses 4.0.3 (checked on macOS and Windows) because the only dependency reference I could find is [this one](https://github.com/FreeCAD/FreeCAD-LibPack/commit/6857c49631efcb0a71ae636197d3da46d4ea2fa2#diff-587cb980af76fdc7e52369fd0b9d926dff266976b6f8ac631e358fecc49ff8cfL111-L113) where 4.0.8 is referenced.

stelioskat, 2026-09-18:
I really hope it will soon be fixed! Are there any news or plans from the devs to address this bug?

dhoepfl, 2026-10-31:
It seems like this has been fixed in 26.3RC2.

closed 2026-10-31 by dhoepfl