FreeCAD/FreeCAD #20304: Deleting result fo Persistend Section Cut crashes freecad

leoheck, 2025-03-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Still cannot delete a Part Boolean operation without crashing Freecad.

Why is that?

Why it cannot delete things recursively?
Is there any other way of deleting things?

I am doing a `Persistent Section Cut`, and then Deleting the resulting model.

Delete
[image]

Delete
[image]

Delete
[image]

```
Opening in existing browser session.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x7000c4645330]
#1  0x7000c5cfa9a6 in QDialogButtonBox::button(QDialogButtonBox::StandardButton) const from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5+0x6
#2  0x7000934b0307 in non-virtual thunk to Gui::TaskMeasure::onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAgflbda/usr/lib/MeasureGui.so+0xa7
#3  0x7000c7f73a5c in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libFreeCADGui.so+0x2c
#4  /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libFreeCADGui.so(+0x806a2d) [0x7000c7c06a2d]
#5  0x7000c7f7565f in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libFreeCADGui.so+0x2ef
#6  /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libFreeCADGui.so(+0x806a2d) [0x7000c7c06a2d]
#7  0x7000c7f7592e in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libFreeCADGui.so+0x28e
#8  0x7000c7f77be4 in Gui::SelectionSingleton::addSelection(char const*, char const*, char const*, float, float, float, std::vector<Gui::SelectionSingleton::SelObj, std::allocator<Gui::SelectionSingleton::SelObj> > const*, bool) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libFreeCADGui.so+0x954
#9  0x7000c7bf7c3d in Gui::DocumentItem::updateItemSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libFreeCADGui.so+0x49d
#10  0x7000c7bf7f18 in Gui::DocumentItem::updateSelection(QTreeWidgetItem*, bool) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libFreeCADGui.so+0x88
#11  0x7000c7c221d6 in Gui::TreeWidget::onItemSelectionChanged() from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libFreeCADGui.so+0x326
#12  /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7000c4ceb6f2]
#13  /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5(+0x43cc76) [0x7000c5e3cc76]
#14  /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x7000c4ceb964]
#15  0x7000c4c64ce7 in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Core.so.5+0x27
#16  0x7000c4c6b293 in QItemSelectionModel::emitSelectionChanged(QItemSelection const&, QItemSelection const&) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Core.so.5+0x2d3
#17  0x7000c4c6d08e in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Core.so.5+0x1ee
#18  0x7000c5e318e8 in QTreeViewPrivate::select(QModelIndex const&, QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5+0x688
#19  0x7000c5e321b7 in QTreeView::setSelection(QRect const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5+0x167
#20  0x7000c5dc36b7 in QAbstractItemView::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5+0x577
#21  0x7000c5baec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5+0x178
#22  0x7000c5c56c6e in QFrame::event(QEvent*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5+0x1e
#23  0x7000c4cb68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Core.so.5+0x7b
#24  0x7000c5b7039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#25  0x7000c5b78262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#26  0x7000c797e2f9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libFreeCADGui.so+0xc9
#27  0x7000c4cb6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Core.so.5+0x108
#28  0x7000c5b763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#29  /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7000c5bc6e6f]
#30  /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7000c5bca0c9]
#31  0x7000c5b703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#32  0x7000c797e2f9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libFreeCADGui.so+0xc9
#33  0x7000c4cb6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Core.so.5+0x108
#34  0x7000c53353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Gui.so.5+0x84c
#35  0x7000c530b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Gui.so.5+0xac
#36  /tmp/.mount_FreeCAgflbda/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7000bf4a2bea]
#37  /tmp/.mount_FreeCAgflbda/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7000c3310ad7]
#38  /tmp/.mount_FreeCAgflbda/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7000c3313ac7]
#39  /tmp/.mount_FreeCAgflbda/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7000c33140ce]
#40  0x7000c4d0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Core.so.5+0x66
#41  0x7000c4cb570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Core.so.5+0x11b
#42  0x7000c4cbd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libQt5Core.so.5+0x82
#43  0x7000c78dd80d in Gui::Application::runApplication() from /tmp/.mount_FreeCAgflbda/usr/bin/../lib/libFreeCADGui.so+0x70d
#44  /tmp/.mount_FreeCAgflbda/usr/bin/freecad(+0x7906) [0x55c2b77d5906]
#45  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7000c462a1ca]
#46  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7000c462a28b]
#47  /tmp/.mount_FreeCAgflbda/usr/bin/freecad(+0x7c8e) [0x55c2b77d5c8e]
```

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * fasteners.backup1742217927.3036284 0.5.37 (Disabled)
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.21
  * DynamicData 2.73.0
  * Curves 0.6.59
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

hyarion, 2025-03-19:
Can you provide a test file and step-by-step instructions on how to reproduce the crash?

ddund, 2025-03-22:
Hi,

I think have the same issue or maybe it is related at least. I have a file where I am trying to edit a previous feature in the tree.

<details>

<summary>My backtrace:</summary>

```Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x3ee80) [0x7f7469a51e80]
#1  0x7f73f041582e in TopLoc_Location::Multiplied(TopLoc_Location const&) const from /app/lib/libTKMath.so.7.8+0x1e
#2  0x7f73f1b07039 in BRep_Tool::Surface(TopoDS_Face const&, TopLoc_Location&) from /app/lib/libTKBRep.so.7.8+0x39
#3  0x7f73f1b1c5fb in BRepAdaptor_Surface::Initialize(TopoDS_Face const&, bool) from /app/lib/libTKBRep.so.7.8+0xab
#4  0x7f73edf63802 in ChFi3d_Builder::StartSol(opencascade::handle<ChFiDS_Spine> const&, opencascade::handle<BRepAdaptor_Surface>&, gp_Pnt2d&, opencascade::handle<BRepAdaptor_Curve2d>&, double&, opencascade::handle<ChFiDS_SurfData> const&, bool, int, opencascade::handle<BRepAdaptor_Surface>&, opencascade::handle<BRepAdaptor_Curve2d>&, bool&, bool&, bool&, bool&, opencascade::handle<BRepAdaptor_Surface>&, gp_Pnt2d&, bool, TopoDS_Vertex const&) const from /app/lib/libTKFillet.so.7.8+0x8c2
#5  0x7f73edf66a7c in ChFi3d_Builder::PerformSetOfSurfOnElSpine(opencascade::handle<ChFiDS_ElSpine> const&, opencascade::handle<ChFiDS_Stripe>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, bool) from /app/lib/libTKFillet.so.7.8+0xa5c
#6  0x7f73edf6b048 in ChFi3d_Builder::PerformSetOfKGen(opencascade::handle<ChFiDS_Stripe>&, bool) from /app/lib/libTKFillet.so.7.8+0xc8
#7  0x7f73edf6ddf2 in ChFi3d_Builder::PerformSetOfSurf(opencascade::handle<ChFiDS_Stripe>&, bool) from /app/lib/libTKFillet.so.7.8+0x72
#8  0x7f73edf36286 in ChFi3d_Builder::Compute() from /app/lib/libTKFillet.so.7.8+0x4d6
#9  0x7f73ee0abf43 in BRepFilletAPI_MakeFillet::Build(Message_ProgressRange const&) from /app/lib/libTKFillet.so.7.8+0x33
#10  0x7f73f1069aec in BRepBuilderAPI_MakeShape::Shape() from /app/lib/libTKTopAlgo.so.7.8+0x8c
#11  0x7f73f17b47c0 in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /app/freecad/lib/Part.so+0x180
#12  0x7f73f17b48c6 in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, Part::TopoShape const&, char const*) from /app/freecad/lib/Part.so+0x86
#13  0x7f73f17b60a5 in Part::TopoShape::makeElementFillet(Part::TopoShape const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, double, double, char const*) from /app/freecad/lib/Part.so+0x125
#14  0x7f73e218ed26 in PartDesign::Fillet::execute() from /app/freecad/lib/_PartDesign.so+0x276
#15  0x7f746c73522a in App::DocumentObject::recompute() from /app/freecad/lib/libFreeCADApp.so+0x9a
#16  0x7f73f15452c0 in Part::Feature::recompute() from /app/freecad/lib/Part.so+0x10
#17  0x7f746c6c0b44 in App::Document::_recomputeFeature(App::DocumentObject*) from /app/freecad/lib/libFreeCADApp.so+0x334
#18  0x7f746c6c7e19 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /app/freecad/lib/libFreeCADApp.so+0x5e9
#19  0x7f746c7a017f in App::DocumentPy::recompute(_object*) from /app/freecad/lib/libFreeCADApp.so+0x2ef
#20  0x7f746c7a02a8 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /app/freecad/lib/libFreeCADApp.so+0x28
#21  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c3116) [0x7f746bbc3116]
#22  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x97) [0x7f746bbc2f17]
#23  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x2db6) [0x7f746bb06196]
#24  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2d2176) [0x7f746bcd2176]
#25  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyEval_EvalCode+0xaa) [0x7f746bcd222a]
#26  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2d2b24) [0x7f746bcd2b24]
#27  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2d2c1b) [0x7f746bcd2c1b]
#28  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyRun_StringFlags+0x74) [0x7f746bcd2ed4]
#29  0x7f746c16e92e in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /app/freecad/lib/libFreeCADBase.so+0x6e
#30  0x7f746d3902aa in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /app/freecad/lib/libFreeCADGui.so+0x9a
#31  0x7f746d3ad1c2 in StdCmdDelete::activated(int) from /app/freecad/lib/libFreeCADGui.so+0x1052
#32  0x7f746d3957e7 in Gui::Command::_invoke(int, bool) from /app/freecad/lib/libFreeCADGui.so+0x237
#33  0x7f746d395c86 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /app/freecad/lib/libFreeCADGui.so+0x136
#34  /usr/lib/x86_64-linux-gnu/libQt6Core.so.6(+0x1e55fe) [0x7f746a1e55fe]
#35  0x7f746ae09004 in QAction::triggered(bool) from /usr/lib/x86_64-linux-gnu/libQt6Gui.so.6+0x44
#36  0x7f746ae0bfbb in QAction::activate(QAction::ActionEvent) from /usr/lib/x86_64-linux-gnu/libQt6Gui.so.6+0x10b
#37  0x7f746d405e18 in Gui::ShortcutManager::onTimer() from /app/freecad/lib/libFreeCADGui.so+0xf8
#38  0x7f746d407100 in Gui::ShortcutManager::checkShortcut(QObject*, QKeySequence const&) from /app/freecad/lib/libFreeCADGui.so+0x620
#39  0x7f746d4080d3 in Gui::ShortcutManager::eventFilter(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x83
[... cut ...]
#64  FreeCAD(+0x9607) [0x560bdd12d607]
#65  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x2808a) [0x7f7469a3b08a]
#66  /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7f7469a3b14b]
#67  FreeCAD(+0x9955) [0x560bdd12d955]
```
</details>

Here is the file that produces the crash: [segfault-crash.zip](https://github.com/user-attachments/files/19402086/segfault-crash.zip) When I try to delete Fillet005 it crashes. I've tried selecting other edges before deleting but that does not work either.

hyarion, 2025-03-22:
> I think have the same issue or maybe it is related at least. I have a file where I am trying to edit a previous feature in the tree.

@ddund it crashes in a new place, could you create a new issue for it, please?