FreeCAD/FreeCAD #20315: Strange symbols in the quick access panel in the file save/open menu. Cyrillic characters in the paths.

kaiwas, 2025-03-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I don't know what exactly this problem is related to, but the menu for opening or saving a file on my computer looks like this if I have English set as the default language.

[image]

If I set Russian as the default language, everything looks fine.

[image]

The problem is probably related to Cyrillic characters in the paths.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Test
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * Road 2025.2.22
  * addFC 1.4.3
  * Curves 0.6.60
  * freecad.gears 1.3.0
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-20:
Do you alter the FreeCAD or OS language?
Maybe related: https://github.com/FreeCAD/FreeCAD/issues/12006

kaiwas, 2025-03-21:
> Do you alter the FreeCAD or OS language?

FreeCAD

The problem is found only in this menu.

kaiwas, 2025-03-25:
@maxwxyz  The issue has been resolved in build 40971.
You can close it.
Thank you!

[image]

closed 2025-03-25 by kaiwas

kaiwas, 2025-11-21:
UPD/ I deleted the problematic paths and replaced them with new ones (by dragging them from the right side).
Perhaps the problematic paths were inherited from settings in previous versions.

I've asked my colleagues to confirm or deny this issue. Perhaps there really is no problem, and I'll temporarily close the issue.

[image]