FreeCAD/FreeCAD #20323: estlcam postprocessor file saving issues

Duffsman, 2025-03-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Can not save gcode generated with eytlcam processor, grbl for example works fine. 
This is the error I get 
```
command 'CAM_Post' failed:
Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Post/Command.py", line 188, in Activated
    post_data = postprocessor.export()
                ^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Post/Processor.py", line 321, in export
    gcode = self.script_module.export(sublist, "-", self._job.PostProcessorArgs)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Post/scripts/estlcam_post.py", line 331, in export
    gfile = pyopen(filename, "w")
            ^^^^^^^^^^^^^^^^^^^^^

[Errno 30] Read-only file system: '-'

```

So  to fix it I replaced the following part, starting at line 331 of the estlcam_post.py 
```
    # write the file
    gfile = pyopen(filename, "w")
    gfile.write(final)
    gfile.close()
```
with what I found in the grbl_post.py 
```
if filename != "-":
        with pyopen(filename, "w") as gfile:
            gfile.write(final)
```

now it's working for me.

### Full version info

```shell
OS: macOS 15.3.2
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

phaseloop, 2025-03-20:
What postprocessor setting are you using? My best guess that some postprocessor parameter has an extra space or was left as "-" and it's being interpreted as a file name.

Duffsman, 2025-03-20:
Beside the path for the postprocessor output I did'nt specify any other parameters.

maxwxyz, 2025-03-20:
Could you provide an example file and test if this also happens with the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Duffsman, 2025-03-20:
here is an example file:
[estlcam_bug_example.zip](https://github.com/user-attachments/files/19367685/estlcam_bug_example.zip)

I can't test the dev version. When starting the OS complains, that the application is damaged and can not be executed.

LarryWoestman, 2025-03-20:
There was a change (somewhere around the 21.0 time frame, I think) to add "-" as a file name to mean "don't write out a file" as part of the refactoring to move where the file was written out in the code.
Some of the postprocessors were fixed, but some of the postprocessors don't appear to have an active "maintainer" and were not fixed.
The export function/method is expected to return the gcode.

phaseloop, 2025-03-20:
So you are saying that in ideal situation no postprocessor should write on their own but instead `export()` should only return gcode?

LarryWoestman, 2025-03-20:
The rules changed for postprocessors around the 21.0 time frame.  Before that the postprocessor handled writing out the file, after that the code that calls the postprocessor started to handle writing out the file, and the old-style postprocessors were now passed a file name of "-" telling them that they didn't need to write out the file in that case.
Those postprocessors that were being maintained were changed to work the "new" way, but there are still some "unmaintained" postprocessors around that probably haven't been updated to the newer behavior.
It appears to me that the estlcam postprocessor hasn't been updated.
It is still possible for postprocessors to write out the file themselves (or store the file on a file server, which at least one postprocessor does).  Much more recently I filed a PR that allows a postprocessor to return None where the gcode would normally be returned to indicate to the code that calls the postprocessor that it doesn't need to write out a file.

phaseloop, 2025-03-20:
cool, thanks for explanation!

phaseloop, 2025-03-23:
Fix: https://github.com/FreeCAD/FreeCAD/pull/20407

J-Dunn, 2025-03-25:
> Much more recently I filed a PR that allows a postprocessor to return None where the gcode would normally be returned to indicate to the code that calls the postprocessor that it doesn't need to write out a file.

Could you link that PR.   Not being able to cancel writing a file when inspecting post-proc output is one of the biggest time-wasters I have. 

It's really annoying that pressing cancel and pressing OK in the gcode editor do exactly the same thing: send to the file save dlg. 

I looked at the code on this a while back and because of the way the editor widget was run there was no obvious fix without does some higher up code reshuffling. 

I use GRBL and I'm not seeing any change in this behaviour. 
If you have a solution , I'd like to see it and see the state of that PR. Thanks.

closed 2025-03-25 by hyarion

LarryWoestman, 2025-03-25:
The change was part of https://github.com/FreeCAD/FreeCAD/pull/19006 (in the Command.py file).
I don't think the the GRBL postprocessor code has been changed to use this capability.  So far as I know only one "custom" postprocessor has used it.

J-Dunn, 2025-03-25:
Thanks, 

I just tested with refactored_grbl and did not see any difference. 

Can you indicate what needs doing and I'll see whether I can submit what's needed.

LarryWoestman, 2025-03-26:
For both the grbl and refactored_grbl code, we would need to look at the output of the gcode editor dialog and figure out how to recognize that Cancel was pressed, then return None where the gcode text would normally be returned to the code that calls the postprocessor, but I haven't tried to figure out exactly what would need to change.

LarryWoestman, 2025-03-26:
The gcode editor dialog is called in the grbl post code just before writing out the file.  In the refactored code base the gcode editor is called just before the end of the UtilsExport.py file.

J-Dunn, 2025-03-26:
It's while since I looked at this problem but from memory the gcode editor is a standard qt widget which returns binary value indicating whether is was OKed or Cancelled. 

However, as you noted in the old grbl-post this result was too late to prevent calling the saveAs dlg. That is why there was no easy fix. It required refactoring to calling code. 

Now that sterling work has been done it should simply be a case of testing the result of the exec() which calls the gcode editor.