FreeCAD/FreeCAD #20327: Sketcher: rounded square shell messed up when increasing distance from axis.

RiccaDS, 2025-03-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the file I attach, when you increase the distance of the geometry from the Y axis from 10 to 400mm, the geometry gets messed up. I often find this problem. 

### Full version info

```shell
OS: Ubuntu 24.10 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/140.99/1
Installed mods: 
  * Curves 0.6.55
  * Manipulator 1.5.7
  * Estimate 0.1.2
  * fasteners 0.5.37
  * ThreadProfile 1.92.0
  * Defeaturing 1.2.3
  * A2plus 0.4.68
```

### Subproject(s) affected?

None

### Anything else?

[messedsquare.zip](https://github.com/user-attachments/files/19369068/messedsquare.zip)

[image]

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-20:
This is a known issue as there are multiple solutions to the constraints

closed 2025-03-20 by maxwxyz

kaiwas, 2025-04-10:
> This is a known issue as there are multiple solutions to the constraints

There will be hundreds more like these, because the existing solutions to this problem are not at all obvious.