FreeCAD/FreeCAD #20329: Crash when using "Measure" tool in Parts Design wb

ebolisa, 2025-03-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The program crashes when measuring an object. It does not happen with released version 1.0 is.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Spanish/Spain (es_ES)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/112.889
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.12.6
  * Curves 0.6.8
  * dodo 1.0.0
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Lithophane
  * sheetmetal 0.7.21
  * ThreadProfile 1.84.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-03-20:
Did you check to see if bug exists already ? https://github.com/FreeCAD/FreeCAD/issues?q=state%3Aopen%20label%3A%22Mod%3A%20Measurement%22

Can you zip up an example file and drag/drop it into Github textbox ?

ebolisa, 2025-03-20:
> Did you check to see if bug exists already ? https://github.com/FreeCAD/FreeCAD/issues?q=state%3Aopen%20label%3A%22Mod%3A%20Measurement%22
> 
> Can you zip up an example file and drag/drop it into Github textbox ?

Sure. I noticed it crashes when "Length" is selected and not when "Distance" is selected.

[temp.zip](https://github.com/user-attachments/files/19375573/temp.zip)

kb528396, 2025-03-23:
For me, it happens when I press reset and start a second measurement. Auto is on and it auto selects distance and crashes or gets stuck on the measurement task panel every time. From there you cant save or close the task panel.

ezequielleonzybert, 2025-03-24:
This happens to me too. I noticed that in the 1.1 weekly build, the tool resets the metrics when you click on a blank space, whereas in the 1.0 build, it didn't. Could this be related to this bug?

Giraut, 2025-06-15:
It happens to me too. I can confirm this in 1.1.0dev Linux amd64 and arm64.

More specifically, my experience is that FreeCAD crashes when the item(s) selected for measurement are deselected  - i.e. when clicking on the side to deselect them, pressing ESC or clicking on Reset.

However, I found a workaround: in the Measurement settings (small icon on top of the Measurement task tab), if Additive Selection is enabled, FreeCAD doesn't crash anymore.

luzpaz, 2025-06-16:
@Giraut can you please add you full [About](https://wiki.freecad.org/About) info for each instance you tested on ?

Syres916, 2025-06-16:
I've tested https://github.com/FreeCAD/FreeCAD/pull/22016 with regrd to another report on the forum and I cannot get a crash. The title regarding _PartDesign_ is irrelevant to me, it's any 3D object IMHO.

Giraut, 2025-06-16:
> [@Giraut](https://github.com/Giraut) can you please add you full [About](https://wiki.freecad.org/About) info for each instance you tested on ?

Sure:

- This is my x86 machine:

OS: Linux Mint 22 (i3/i3/xcb)
Architecture: x86_64
Version: 1.1.0dev.42079 (Git)
Build date: 2025/06/05 12:02:25
Build type: Release
Branch: main
Hash: de37c5157aeb134d64bf76cd9d5bb3d644b39c5a
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96.1263/158.196/1
Installed mods: 
  * Assembly3 0.12.2
  * kicadStepUpMod 11.2.9
  * Launcher
  * Assembly4 0.50.7

- This is my ARM64

OS: Debian GNU/Linux 13 (trixie) (sway/sway/wayland)
Architecture: arm64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/141.471/1.25

m1cha1, 2025-06-17:
I can confirm that it is not working properly. The problem occurs when two vertices are selected:
> 09:16:00  <Exception> Access violation
> 09:16:00  0 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 

Tested also in safe mode.

My configuraion:
> OS: Windows 10 build 19045
> Architecture: x86_64
> Version: 1.1.0dev.42180 (Git)
> Build date: 2025/06/16 16:25:59
> Build type: Release
> Branch: main
> Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
> Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
> shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
> Locale: Polish/Poland (pl_PL)
> Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
> Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
> Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1
>

xtemp09, 2025-06-17:
_FreeCAD_ crashes if I select two vertices with the measure tool. If I select and deselect shows warning about unhandled exception.

Below is the backtrace and the about info.

<details>

<details><summary>gdb</summary>

```cpp
<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
54.6653 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 
<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
90.4073 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 
<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
113.246 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 

Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
0x00005555555844f2 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_data (this=0x0) at /usr/include/c++/13/bits/basic_string.h:223
223           { return _M_dataplus._M_p; }
(gdb) bt
#0  0x00005555555844f2 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_data (this=0x0) at /usr/include/c++/13/bits/basic_string.h:223
#1  0x0000555555582346 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::c_str (this=0x0) at /usr/include/c++/13/bits/basic_string.h:2596
#2  0x00007fffccf1362c in Measure::MeasurePosition::execute (this=0x55555a658750)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Measure/App/MeasurePosition.cpp:99
#3  0x00007ffff378e224 in App::DocumentObject::recompute (this=0x55555a658750)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/DocumentObject.cpp:168
#4  0x00007fffccf13862 in Measure::MeasurePosition::onChanged (this=0x55555a658750, 
    prop=0x55555a658cf0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Measure/App/MeasurePosition.cpp:119
#5  0x00007ffff3975f24 in App::Property::hasSetValue (this=0x55555a658cf0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/Property.cpp:274
#6  0x00007ffff39a90ef in App::PropertyLinkBase::hasSetValue (this=0x55555a658cf0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/PropertyLinks.cpp:95
#7  0x00007ffff39b14c2 in App::PropertyLinkSub::setValue (this=0x55555a658cf0, 
    lValue=0x0, subs=..., shadows=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/PropertyLinks.cpp:1359
#8  0x00007ffff39b6e53 in App::PropertyLinkSub::breakLink (this=0x55555a658cf0, 
    obj=0x55555a658750, clear=true)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/PropertyLinks.cpp:2087
#9  0x00007ffff39bf610 in App::PropertyLinkBase::breakLinks (link=0x55555a658750, 
    objs=std::vector of length 3, capacity 4 = {...}, clear=true)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/PropertyLinks.cpp:3708
#10 0x00007ffff36279ca in App::Document::breakDependency (this=0x55555798b000, 
    pcObject=0x55555a658750, clear=true)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/Document.cpp:3632
#11 0x00007ffff362754a in App::Document::_removeObject (this=0x55555798b000, 
    pcObject=0x55555a658750, options=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/Document.cpp:3592
#12 0x00007ffff3626b4b in App::Document::removeObject (this=0x55555798b000, 
    sName=0x555556e642e8 "Position")
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/Document.cpp:3523
#13 0x00007fffa95acbd0 in Gui::TaskMeasure::removeObject (this=0x55555a4a25d0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Measure/Gui/TaskMeasure.cpp:439
#14 0x00007fffa95ac3b0 in Gui::TaskMeasure::update (this=0x55555a4a25d0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Measure/Gui/TaskMeasure.cpp:298
#15 0x00007fffa95ace1f in Gui::TaskMeasure::onSelectionChanged (this=0x55555a4a25d0, 
    msg=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Measure/Gui/TaskMeasure.cpp:478
#16 0x00007ffff600d3b5 in Gui::SelectionObserver::_onSelectionChanged (
    this=0x55555a4a2640, msg=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/Selection/Selection.cpp:155
#17 0x00007ffff604d364 in std::__invoke_impl<void, void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&> (
    __f=@0x5555579c1f30: (void (Gui::SelectionObserver::*)(Gui::SelectionObserver * const, const Gui::SelectionChanges &)) 0x7ffff600d34c <Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&)>, __t=@0x5555579c1f40: 0x55555a4a2640)
    at /usr/include/c++/13/bits/invoke.h:74
#18 0x00007ffff604cb95 in std::__invoke<void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&> (
    __fn=@0x5555579c1f30: (void (Gui::SelectionObserver::*)(Gui::SelectionObserver * const, const Gui::SelectionChanges &)) 0x7ffff600d34c <Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&)>) at /usr/include/c++/13/bits/invoke.h:96
#19 0x00007ffff604c0ee in std::_Bind<void (Gui::SelectionObserver::*(Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>::__call<void, Gui::SelectionChanges const&, 0ul, 1ul>(std::tuple<Gui::SelectionChanges const&>&&, std::_Index_tuple<0ul, 1ul>) (this=0x5555579c1f30, __args=...) at /usr/include/c++/13/functional:506
[... cut ...]
Build date: 2025/06/05 12:39:32
Build type: Unknown
Branch: master
Hash: 5c219bea4187030d56c998bc318bdf4869209ced
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96.2463/96.2463/1
```

</details>

GAZ082, 2025-06-20:
Confirmed. Annoying. :)

OS: Arch Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.42180 (Git) AppImage
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Object center
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/101.6/1
Installed mods: 
  * fasteners 0.5.38
  * OpenTheme 2024.9.1

PR #22016 (2025-06-23): [Measure] Fix seg fault in MeasurePosition::execute if subElements is empty
This PR is to fix the seg fault reported on the [forum](https://forum.freecad.org/viewtopic.php?t=97756) only when Measurement is started before selection of vertices.
Regression tested pre-selection of vertices with PR successfully.

Fixes #20329 
Fixes #20194 

@hlorus FYI
closed 2025-06-23 by chennes