FreeCAD/FreeCAD #20331: OpenBSD: SIGABRT when closing Preferences dialog

thyssentishman, 2025-03-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I was able to reproduce the crash in two ways:

1. Edit -> Preferences
2. Cancel
3. SIGABRT

<details>
<summary>
Backtrace
</summary>
<pre>
#1  0x2b3494a8c5fcf0c0 in ?? ()
#2  0x000004ce3c38575b in _libc_abort () at /usr/src/lib/libc/stdlib/abort.c:51
#3  0x000004ce3c37abe4 in wrterror (d=0x4cefbbc17e8, msg=0x4ce3c2e6f82 "write to free mem %p[%zu..%zu]@%zu")
    at /usr/src/lib/libc/stdlib/malloc.c:378
#4  0x000004ce3c37c763 in validate_junk (pool=0x4cefbbc17e8, p=0x0, argsz=0) at /usr/src/lib/libc/stdlib/malloc.c:780
#5  ofree (argpool=0x7c521af65b60, p=0x0, clear=<optimized out>, check=<optimized out>, argsz=<optimized out>)
    at /usr/src/lib/libc/stdlib/malloc.c:1716
#6  0x000004ce3c37be33 in _libc_free (ptr=0x4cf190ae930) at /usr/src/lib/libc/stdlib/malloc.c:1747
#7  0x000004ce6c56e68b in operator delete (ptr=0x0) at /usr/src/gnu/lib/libcxx/../../../gnu/llvm/libcxx/src/new.cpp:133
#8  0x000004cf25067902 in QRenderRule::~QRenderRule() () from /usr/local/lib/libQt6Widgets.so.4.0
#9  0x000004cf2506ee20 in QHash<unsigned long long, QRenderRule>::~QHash() () from /usr/local/lib/libQt6Widgets.so.4.0
#10 0x000004cf2506ecdc in QHash<int, QHash<unsigned long long, QRenderRule> >::~QHash() () from /usr/local/lib/libQt6Widgets.so.4.0
#11 0x000004cf2507ccda in QHashPrivate::Data<QHashPrivate::Node<QObject const*, QHash<int, QHash<unsigned long long, QRenderRule> > > >::erase(QHashPrivate::Data<QHashPrivate::Node<QObject const*, QHash<int, QHash<unsigned long long, QRenderRule> > > >::Bucket) ()
   from /usr/local/lib/libQt6Widgets.so.4.0
#12 0x000004cf2507cc54 in bool QHash<QObject const*, QHash<int, QHash<unsigned long long, QRenderRule> > >::removeImpl<QObject const*>(QObject const* const&) () from /usr/local/lib/libQt6Widgets.so.4.0
#13 0x000004cf250538fc in QStyleSheetStyleCaches::objectDestroyed(QObject*) () from /usr/local/lib/libQt6Widgets.so.4.0
#14 0x000004ce4e892b17 in void doActivate<false>(QObject*, int, void**) () from /usr/local/lib/libQt6Core.so.6.0
#15 0x000004ce4e885344 in QObject::destroyed(QObject*) () from /usr/local/lib/libQt6Core.so.6.0
#16 0x000004cf24fe1859 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.4.0
#17 0x000004ce8e0afd31 in Gui::Dialog::PreferencePagePython::~PreferencePagePython (this=0x4ced2ef37c0)
    at /usr/ports/pobj/freecad-1.0.0/FreeCAD-1.0.0/src/Gui/WidgetFactory.cpp:298
#18 0x000004ce8e0afdb3 in Gui::Dialog::PreferencePagePython::~PreferencePagePython (this=0x0)
    at /usr/ports/pobj/freecad-1.0.0/FreeCAD-1.0.0/src/Gui/WidgetFactory.cpp:295
#19 0x000004ce4e885404 in QObjectPrivate::deleteChildren() () from /usr/local/lib/libQt6Core.so.6.0
#20 0x000004cf24fe18a9 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.4.0
#21 0x000004cf251beea3 in QStackedWidget::~QStackedWidget() () from /usr/local/lib/libQt6Widgets.so.4.0
#22 0x000004ce4e885404 in QObjectPrivate::deleteChildren() () from /usr/local/lib/libQt6Core.so.6.0
#23 0x000004cf24fe18a9 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.4.0
#24 0x000004cf251beea3 in QStackedWidget::~QStackedWidget() () from /usr/local/lib/libQt6Widgets.so.4.0
#25 0x000004ce4e885404 in QObjectPrivate::deleteChildren() () from /usr/local/lib/libQt6Core.so.6.0
#26 0x000004cf24fe18a9 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.4.0
#27 0x000004cf24fe24b3 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.4.0
#28 0x000004ce4e885404 in QObjectPrivate::deleteChildren() () from /usr/local/lib/libQt6Core.so.6.0
#29 0x000004cf24fe18a9 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.4.0
#30 0x000004cf24fe24b3 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.4.0
#31 0x000004ce4e885404 in QObjectPrivate::deleteChildren() () from /usr/local/lib/libQt6Core.so.6.0
#32 0x000004cf24fe18a9 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.4.0
#33 0x000004cf251a4da3 in QScrollArea::~QScrollArea() () from /usr/local/lib/libQt6Widgets.so.4.0
#34 0x000004ce4e885404 in QObjectPrivate::deleteChildren() () from /usr/local/lib/libQt6Core.so.6.0
#35 0x000004cf24fe18a9 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.4.0
#36 0x000004cf2508ffd3 in QFrame::~QFrame() () from /usr/local/lib/libQt6Widgets.so.4.0
#37 0x000004ce4e885404 in QObjectPrivate::deleteChildren() () from /usr/local/lib/libQt6Core.so.6.0
#38 0x000004cf24fe18a9 in QWidget::~QWidget() () from /usr/local/lib/libQt6Widgets.so.4.0
#39 0x000004ce8dc2d071 in StdCmdDlgPreferences::activated (this=<optimized out>, iMsg=<optimized out>)
    at /usr/ports/pobj/freecad-1.0.0/FreeCAD-1.0.0/src/Gui/CommandStd.cpp:438
#40 0x000004ce8dbfdbe8 in Gui::Command::_invoke (this=0x4cf14733f00, id=0, disablelog=<optimized out>)
    at /usr/ports/pobj/freecad-1.0.0/FreeCAD-1.0.0/src/Gui/Command.cpp:444
#41 0x000004ce8dbfd94b in Gui::Command::invoke (this=0x4cf14733f00, i=0, trigger=<optimized out>)
    at /usr/ports/pobj/freecad-1.0.0/FreeCAD-1.0.0/src/Gui/Command.cpp:408
#42 0x000004ce4e892b17 in void doActivate<false>(QObject*, int, void**) () from /usr/local/lib/libQt6Core.so.6.0
#43 0x000004ce5b05352f in QAction::activate(QAction::ActionEvent) () from /usr/local/lib/libQt6Gui.so.8.0
#44 0x000004cf25185ddb in QMenuPrivate::activateCausedStack(QList<QPointer<QWidget> > const&, QAction*, QAction::ActionEvent, bool) ()
   from /usr/local/lib/libQt6Widgets.so.4.0
[... 138 lines cut ...]
```
rm -r .config/FreeCAD/ .local/share/FreeCAD/ .cache/FreeCAD/
```

### Full version info

```shell
OS: OpenBSD 7.7 (xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.9, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-03-20:
Can you test to see if you can reproduce on 1.1dev ? (aka HEAD)

luzpaz, 2025-03-22:
@adrianinsaval any ideas ?

thyssentishman, 2025-03-24:
Could this be related to https://github.com/FreeCAD/FreeCAD/issues/16905? From the backtraces it can be seen that exceptions are not caught and propagate through Qt's event handling mechanism. One of the backtraces even shows that it reaches `GUIApplication::notify` where the exception can not be caught any more as describe in the linked issue.

xtemp09, 2025-03-27:
> Could this be related to [#16905](https://github.com/FreeCAD/FreeCAD/issues/16905)? From the backtraces it can be seen that exceptions are not caught and propagate through Qt's event handling [...]

This is possible, however, the backtraces provided show no such thing. To find out, one needs to compile _FreeCAD_ with a debug version of _Qt_.

----

What is the version of _clang_ used to compile _FreeCAD_? Is it the same version used to compile _Qt_?

thyssentishman, 2025-03-27:
> What is the version of _clang_ used to compile _FreeCAD_? Is it the same version used to compile _Qt_?

I've been using clang version 16.0.6. Some Qt6 ports on OpenBSD however have already been migrated to compile with clang 19. These are `pyside-{,tools}`, `shiboken` and `qt6-qttools`.

Should I attempt to build FreeCAD with clang 19?

xtemp09, 2025-03-27:
> Should I attempt to build FreeCAD with clang 19?

It's better to build with a debug version of _Qt_since the crash occurs in the deep of _Qt_.

There is a little chance that compilation with _ASAN_ can shed a light on this.

thyssentishman, 2025-08-14:
I just built FreeCAD from commit b3e11ecd1175a911270fb0c7ab27f43262838918 and the crash is still happening. In addition, I get the following messages in the console when starting FreeCAD which contain errors related to the AddonManager. Not sure if this could be related.

<details>
<summary>Console</summary>
<pre>
FreeCAD 1.0.1, Libs: 1.0.1R39285 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

During initialization the error "No module named 'addonmanager_installation_manifest'" occurred in /usr/local/lib/freecad/Mod/AddonManager/./InitGui.py
Look into the log file for further information
Migrating Start Workbench to Start command... done.
Requested non-existent style parameter token 'UserHighlight1Color'.
Requested non-existent style parameter token 'UserHighlight1Color'.
Requested non-existent style parameter token 'UserHighlight1Color'.
Requested non-existent style parameter token 'UserHighlight1Color'.
Unknown command 'Std_AddonMgr'
Requested non-existent style parameter token 'ToolbarButtonsPadding'.
Requested non-existent style parameter token 'ToolbarButtonsPadding'.
Requested non-existent style parameter token 'ToolbarButtonsPadding'.
Requested non-existent style parameter token 'ToolbarButtonsPadding'.
Requested non-existent style parameter token 'UserHighlight1Color'.
Requested non-existent style parameter token 'UserHighlight1Color'.
Requested non-existent style parameter token 'UserHighlight1Color'.
Requested non-existent style parameter token 'UserHighlight1Color'.
Requested non-existent style parameter token 'UserHighlight1Color'.
Requested non-existent style parameter token 'UserHighlight1Color'.
</pre>
</details>

thyssentishman, 2025-08-14:
@luzpaz sure, here it is:

```
OS: OpenBSD 7.7 ()
Architecture: x86_64
Version: 1.1.0dev.42722 (Git)
Build date: 2025/08/14 08:44:49
Build type: Release
Branch: main
Hash: b3e11ecd1175a911270fb0c7ab27f43262838918
Python 3.12.11, Qt 6.8.3, Coin 4.0.4, Vtk 9.4.1, boost 1_87, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/189.401/1
```

thyssentishman, 2026-03-29:
Sorry for taking so long to look at this. I just built the latest release and I can't reproduce this issue anymore, even with the AddonManager installed. I'll ship the AddonManager when updating OpenBSD's port so users can keep testing.