FreeCAD/FreeCAD #20338: Assembly crash damages file

DavidLister, 2025-03-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I was working in the assembly workbench and in the process of deleting a distance joint constaint when FreeCAD crashed. I have tried reproducing the bug, however cannot seem to do it at this time.

What I can repeat though is the loading issue that I have with the file after the crash occurred. 

```
21:50:21  Fatal Error at file /(files)/4-bar.FCStd, line 310, char 27
21:50:21  Invalid Document.xml: unexpected end of input
21:50:21  Fatal Error at file GuiDocument.xml, line 1, char 1
21:50:21  Exception message is: 
21:50:21  invalid document structure
21:50:21  Reading failed from embedded file: GuiDocument.xml (0 bytes, 2 bytes compressed)
```

I'm not well versed in file recover for FreeCAD, so I don't know to what extent the file is damaged, but I can't see anything after opening other than this warning.

The text file attached is the .FCStd file that shows this issue. I hope this helps in either resolving either the file corruption issue or with making file loading more robust. I'll keep trying to reproduce the crash that caused this and will report back on this issue if I figure out a more repeatable way to trigger it.

Thank you!

### Full version info

```shell
OS: NixOS 24.11 (Vicuna) (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.0.0.Unknown
Build type: Release
Python 3.11.11, Qt 5.15.16, Coin 4.0.3, Vtk 9.2.6, OCC 7.6.2
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.37
```

### Subproject(s) affected?

None

### Anything else?

[4-bar.zip](https://github.com/user-attachments/files/19381987/4-bar.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-04-06:
I can recreate the corrupt file.  
I wonder if the file corruption is related to: 
* #18044

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41127 (Git) Conda AppImage
Build date: 2025/04/01 15:11:50
Build type: Release
Branch: main
Hash: db08735e106561f3a11cf08e02fbf5d9e8fbf6b2
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/166.066/1
Installed mods: 
  * fcVM
  * FreeCAD-Ribbon 1.3.17.1
  * SearchBar 1.0.2
  * fcVM.backup1738519851.1523638 (Disabled)
  * Silk 0.1.5
```

luzpaz, 2025-09-01:
@PaddleStroke could you weigh-in ?

PaddleStroke, 2025-09-05:
Many crashes of assembly were fixed in current 1.1. So without clear reproduction steps and confirmation it crashes in 1.1 it's hard to say.