FreeCAD/FreeCAD #20347: [TD]bspline to circle conversion

WandererFan, 2025-03-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This issue was reported in the German subforum here: https://forum.freecad.org/viewtopic.php?t=93552

The problem involves the incorrect interpretation of bsplines as circles.

### Full version info

```shell
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40545 +3 (Git)
Build date: 2025/03/09 23:48:22
Build type: debug
Branch: HeadlessViewCreation20024
Hash: ac4ba822b409c4d1f7eae20b3d502508f3c8df7f
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 115/141.585
Installed mods: 
  * fasteners 0.5.21
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

benj5378, 2025-03-23:
TODO: add tests

WandererFan, 2025-03-24:
Are you volunteering to make tests?  ;)

benj5378, 2025-03-24:
> Are you volunteering to make tests? ;)

😢

I actually find it a bit interesting. This issue #20347, is it the samme bspline circle conversion or a variant? I mean, are there a common test case that can be used?

Though there are many things I want to do, but don't have time for. I could probably start out by fixing all my draft PRs

closed 2025-03-24 by chennes